HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #1: how to get image output from the chip

rundercaster, 2024-02-14:
hi, i get to the last step of the tutorial and i am not sure how you got that image from the chip 
[image]

it works the same as described in tutorial, i am getting output in console 
b'SENSORDPLIB_STATUS_XDMA_FRAME_READY 84'

Any help would be much appreciated
Thanks

kris-himax, 2024-02-15:
Hi @rundercaster ,
  We use SPI to send the image and meta data to PC.
You can get the jpeg image address reference from [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/tflm_fd_fm.c#L241C32-L241C41), and send the image data by SPI reference from [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/tflm_fd_fm.c#L253).
And we use self defined meta data format to send the algorithm result [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/tflm_fd_fm.c#L264).
And the SPI protocol format you can reference from [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/library/spi_ptl/spi_protocol.h#L12).

Finally, if you want to send the image by SPI, you should also add additional hardware.

We will also release another way which sending image by Uart without any additional hardware soon or later.

Thanks,
Kris

wxzed, 2024-02-23:
@kris-himax Could you tell me if there is a place to download your image display tool? Currently, I have also received SENSORDPLIB_STATUS_XDMA_FRAME_READY, but I want to see the actual image. Can I quickly use your display tool to take a look?

kris-himax, 2024-03-04:
Hi @rundercaster @wxzed ,
You can reference himax github [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2?tab=readme-ov-file#send-image-and-meta-data-by-uart).

Thanks,
Kris

closed 2024-03-07 by kris-himax