HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #3: What are the Grove Vision AI V2 connection pins

hpssjellis, 2024-03-12:
@kris-himax 

What are the active pins between the XIAO and the Grove Vision AI V2?

I need to connect a servo and big motor driver to the xiao while it communicates with the Grove Vision AI V2.

My Guess is on the XIAO connect 5V, GND, SDA (D4), SCL(D5) to the equivalent locations on the Grove. Do you have any better suggestions?

[image]

flagyan, 2024-03-13:
Hi @hpssjellis,

In the [sch](https://files.seeedstudio.com/wiki/grove-vision-ai-v2/Grove_Vision_AI_Module_V2_Circuit_Diagram.pdf), the pin definition is as below, is this what you need?
[image]

stevehuang82, 2024-03-13:
Hi @hpssjellis,

Thanks to @flagyan for replying to the Grove Vision AI V2 pin definition. Grove Vision AI V2 can communicate with XIAO through UART/I2C/SPI/GPIO according to your needs. Regarding XIAO's question "Why D3 cannot be used for PWM output", you can ask Seeed Studio.

hpssjellis, 2024-03-13:
@flagyan and @stevehuang82 yeh I guess the Grove pins are just a mirror image of the XIAO, that makes sense. My main question was more about the minimum pins to communicate between the two boards and I basically have it working just using I2C, 3v3 and GND, so 4 pins.

The reason I asked about D3 is that the boards were both plugged together when I had the compile issue so I was wondering if the Grove Vision AI V2 was using that pin for something.  I should probably test it again now that I have separated the two boards.

So far things are going good with my RoboCar. I still have several more steps to go.

[image]

stevehuang82, 2024-03-13:
Hi @hpssjellis ,

    Grove Vision AI V2 pin D3(RST) is connected to the WE2 chip reset pin. Users can use XIAO pin D3 (GPIO4) to reset Grove Vision AI V2 (WE2 chip).
      You can try toggling Grove Vision AI V2 pin D3(RST) (set low and high) and you will see Grove Vision AI V2 reset.

hpssjellis, 2024-03-14:
> Hi @hpssjellis , Grove Vision AI V2 pin D3(RST) is connected to the WE2 chip reset pin. Users can use XIAO pin D3 (GPIO4) to reset Grove Vision AI V2 (WE2 chip). You can try toggling Grove Vision AI V2 pin D3(RST) (set low and high) and you will see Grove Vision AI V2 reset.

Thanks, that explains the issue I had. I can close this.

closed 2024-03-14 by hpssjellis