HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #4: example code for the tf card

flagyan, 2024-03-12:
Hi @kris-himax,

The board has a TF card slot but I can't find the code for it. Is there an example code for it? Thanks.

hpssjellis, 2024-03-12:
Can 2 models be saved on the TF (micro-SD) card and be loaded on the fly?

stevehuang82, 2024-03-13:
Hi @flagyan @hpssjellis ,

We are working on porting the SD card file system. coming soon. Thanks.

flagyan, 2024-03-13:
Hi @stevehuang82,

Thanks a lot. Is there a rough estimation when it will be ready?

stevehuang82, 2024-03-13:
> Hi @stevehuang82,
> 
> Thanks a lot. Is there a rough estimation when it will be ready?

It should be 1 week. Thanks.

stevehuang82, 2024-03-15:
Hi @flagyan,

We have ported the FatFs file system module to the Seeed Grove Vision AI Module V2. There are two scenario applications for reference.
1. [fatfs_mmc_spi](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/fatfs_mmc_spi) : file system read/write.
2. [allon_sensor_tflm_fatfs](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/allon_sensor_tflm_fatfs) : This scenario app can capture images and store them(JPEG) in an SD card.

flagyan, 2024-03-18:
Hi @stevehuang82 ,

I have checked the example code with tf card of 32g(sdhc/fat32, works) and 64g(sdxc/exfat, doesn't work). Thanks a lot.

stevehuang82, 2024-03-18:
Hi @flagyan ,

You can change FF_LBA64 definition to 1 to enable exfat in ~/scenario_app/fatfs_mmc_spi/ffconf.h.

#define FF_LBA64		1
/* This option switches support for 64-bit LBA. (0:Disable or 1:Enable)
/  To enable the 64-bit LBA, also exFAT needs to be enabled. (FF_FS_EXFAT == 1) */

flagyan, 2024-03-19:
Hi @stevehuang82 ,

After setting the FF_LBA64=1 and FF_FS_EXFAT=1, I got the below error on the app allon_sensor_tflm_fatfs with 64g sdxc card:

slot flash_offset 0x00000000
New MemDesp himax_sec_SB_image_process PASS
set_memory_s_ns
bl_status = 0x800000, HX_DSP_FLAG 1
bl_status = 0x800000
jump_addr=0x10000000
Compiler Version: ARM GNU, 13.2.1 20231009
wakeup_event=0x0,WakeupEvt1=0x0
Mount SD card fatfs
cur_dir = 0:/
   <DIR>   System Volume Information
     66048
Entering BusFault_Handler interrupt!
SCB->CFSR:0x00008200
                    SCB->BFAR:0x0099c10a
                                        SCB->HFSR:0x00000000

stevehuang82, 2024-03-20:
Hi @flagyan ,

After setting the "FF_LBA64=1 and FF_FS_EXFAT=1" in ~/allon_sensor_tflm_fatfs/ffconf.h and make clean/make app allon_sensor_tflm_fatfs. The application can capture images and store them in the sdxc(exFAT) card successfully.
The following is my sd card type and format.
[image]
[image]

flagyan, 2024-03-20:
Hi @stevehuang82 ,

I also did the make clean after changing the setting. 

The card I used is Sandisk Ultra 64G [SanDisk Ultra® microSD™ UHS-I Card](https://www.westerndigital.com/en-sg/products/memory-cards/sandisk-ultra-uhs-i-chromebook-microsd?sku=SDSQUAB-064G-GN6FA) and formatted to exfat by SD Memory Card Formatter from sdcard.org, the read/write has no flaw under the windows os.

Anyhow as I can continue my stuff with the sdhc card and the exfat support has no problem in your side. I think we can close the issue as the tf card support is already done.

I will create a new issue if there are any new findings. Thanks a lot.

closed 2024-03-21 by stevehuang82