HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #7: Putting JPEG data to the yolov8_od model

saricaBugrahan, 2024-03-15:
Hi @kris-himax,

I am trying to use the tflm_yolov8_od model with a custom image that is coming from the another mcu. I am able to transfer the data to the device via UART. However, I am sending the data in JPEG format.  I am putting the image data into the model like below: 
<code>for(uint32_t i = 0;i<image_size;i++)
		{
			*((uint8_t *)yolov8n_ob_input->data.data+i) = (uint8_t)response_image[i];
		}
		// //uint8 to int8
		for (int i = 0; i < yolov8n_ob_input->bytes; ++i) {
			*((int8_t *)yolov8n_ob_input->data.data+i) = *((int8_t *)yolov8n_ob_input->data.data+i) - 128;
    	}</code>

At the end, if the image is coming from the camera with RAW data, it can able to find the objects. But If I insert the compressed image into the model, it result with no output. What could be the reason of it or is it possible to use JPEG images on the model ?

Thank you

kris-himax, 2024-03-19:
I think that it is not.

kris-himax, 2024-04-01:
You can reference the original repository of yolov8 https://github.com/ultralytics/ultralytics

closed 2024-04-01 by saricaBugrahan