HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #15: Does the new SCMA library work with the old Grove V2 Models

hpssjellis, 2024-05-24:
**Does the new SCMA library work with the old Grove V2 Models**

I have 2 new grove vision V2 and I have uploaded new models using [sensecraft](https://sensecraft.seeed.cc/ai/#/device/local) and it mentioned updating the firmware when it installed a face detection model.

I used the latest SSCMA library. Every thing works great.

I then tried to compile my old RoboCar code with my first Grove Vision V2 with my custom model. It compiles fine but I don't get any serial monitor reading. I am not 100% sure, but if I replace SSCMA library with an old version things run again for my roboCar.

Does any of these issues make sense? I can probably just reflash my old grove vision V2 model which will auto update the firmware, but before doing that I thought I would just ask if there was a breaking change with the latest version of SSCMA and the Grove firmware.

@flagyan and @stevehuang82

stevehuang82, 2024-05-27:
Hi @hpssjellis ,

From what I understand, your Grove Vision V2 is running the Seeed SSCMA firmware, not the Himax "Seeed_Grove_Vision_AI_Module_V2" repository firmware. is this correct ?

The Himax “Seeed_Grove_Vision_AI_Module_V2” repository firmware provides examples of TensorFlow Lite machine learning models for microcontrollers. It is different from SSCMA (Seeed SenseCraft Model Assistant).

If your Grove Vision V2 is running SSCMA firmware, maybe you can ask Seeed this question to understand the differences between the new/old firmware.

hpssjellis, 2024-05-27:
thanks @stevehuang82 I can close this.

closed 2024-05-27 by hpssjellis