HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #16: SenseCraft AI says latest Firmware is  2024.05.22 but only 2023_12_19 exists here.

KevWal, 2024-06-01:
Hi

The Sensecraft AI Web interface connected to my Grive Vision AI V2 says:

"Device Version:
v2023.12.14

The latest firmware version is detected: 2024.05.22, if you want to upgrade to the latest firmware, you need to select a model to deploy in this platform."

But deplying a model fails and 2024.05.22 doesnt look to exist on this site to enable me to manually upload it?

Thanks
Kevin

stevehuang82, 2024-06-03:
Hi @KevWal ,

    Grove Vision AI module V2 runs SSCMA(Seeed SenseCraft Model Assistant)  firmware by default. If you want to upgrade the firmware, you can connect to the following website.
https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process
or 
https://sensecraft.seeed.cc/ai/#/model

    If updating the firmware fails, you can get help from Seeed. (https://www.seeedstudio.com/get_help/HowToGetHelp)

    The Himax “Seeed_Grove_Vision_AI_Module_V2” repository source code provides examples of TensorFlow Lite machine learning models for microcontrollers. It is different from SSCMA (Seeed SenseCraft Model Assistant) firmware. 

    If you flash the Himax sample application firmware image to your device, the SSCMA firmware will be removed. The device will be able to connect to the Sensecraft AI web interface. In this case, you should flash the firmware image "Seeed_SenseCraft_AI_2023_12_19T183401_402.img"
to reinstall the SSCMA firmware into your device.

KevWal, 2024-06-03:
Hi @stevehuang82 

Thankyou *very* much for taking the time to explain - so the version numbers I quote are both for the SSCMA firmware, but that is generally not what is hosted here in this repo.

The Seeed_SenseCraft_AI_2023_12_19T183401_402.img image on this site is the SSCMA firmware, to enable a user to get back into the SSCMA world rather than the Himax firmware code in this repo.

Just one typo I think in your reply above that may confuse future readers - you say "The device will be able to connect to the Sensecraft AI web interface." I think you mean will *not* be able to connect to Sensecraft web interface [if running this repo's code].

You have obviously been able to get hold of Seeed_SenseCraft_AI_2023_12_19T183401_402.img to host it here, I wonder if you could also host 2024.05.22 to give people a different avenue to get the latest firmware?

Thanks very much
Kevin

stevehuang82, 2024-06-03:
Hi @KevWal 

Thanks for correcting the typo, you are right "If you flash the Himax sample application firmware image to your device, the SSCMA firmware will be removed. The device will NOT be able to connect to the Sensecraft AI web interface."

When users deploy AI models from the Seeed Sensecraft AI website, the website will install the AI model and the latest SSCMA firmware to the device. Therefore, we do not need to keep the latest SSCMA firmware in this repository.

Thanks very much