HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #18: Error running app tflm_fd_fm

hatrinh-sj, 2024-06-10:
I compiled the app tflm_fd_fm successfully. However, I get an error in the serial console.

```
Ethos-U55 device initialised
[ERROR] FD model's schema version -1 is not equal to supported version 3
cis_OV5647_init
```

It appears that the FD model is missing. Please guide me on how to fix it.

kris-himax, 2024-06-11:
Hi @hatrinh-sj ,

You should burn the multiple models by the following command.
```
python xmodem\xmodem_send.py --port=[your COM number] --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\tflm_fd_fm\0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo\tflm_fd_fm\1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo\tflm_fd_fm\2_il_0x32A000.tflite 0x32A000 0x00000"

# example:
# python xmodem\xmodem_send.py --port=COM123 --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\tflm_fd_fm\0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo\tflm_fd_fm\1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo\tflm_fd_fm\2_il_0x32A000.tflite 0x32A000 0x00000"

```
You can reference by the [How to build face mesh scenario app exmple and run on WE2?](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2?tab=readme-ov-file#windows-environment-1).

stevehuang82, 2024-06-11:
Hi @hatrinh-sj 

Please refer to the following web page for instructions. You can using python scripts to program the FW image and models into the flash through UART.

https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2?tab=readme-ov-file#windows-environment-1

 python xmodem\xmodem_send.py --port=[your COM number] --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\tflm_fd_fm\0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo\tflm_fd_fm\1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo\tflm_fd_fm\2_il_0x32A000.tflite 0x32A000 0x00000"

hatrinh-sj, 2024-06-12:
Thank you very much. I missed that part. It works now.

closed 2024-06-12 by hatrinh-sj