HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #19: Memory map

TCHeish, 2024-06-11:
Is there any documentation about memory map like [this](https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ml-embedded-evaluation-kit/+/refs/heads/main/docs/sections/appendix.md#appendix). Thanks.

stevehuang82, 2024-06-11:
Hi @TCHeish ,

You can find the memory map here. https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/FAQ.md

TCHeish, 2024-06-11:
Hi @stevehuang82 
Thanks. How about FLASH? Is it the sec_6 mention [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/342824ab1772705fba5977ee16d95fc193f4bf0a/we2_image_gen_local/common/sec_mem/sec_mem_layout_one_sonly_bin.json#L27) ?

stevehuang82, 2024-06-12:
Hi @TCHeish 
Yes, the starting address of Flash in XIP mode is 0x3A000000.

closed 2024-06-13 by stevehuang82