HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #20: Model location - FLASH_XIP_MODEL

TCHeish, 2024-06-11:
```
 *	1: model file will off-line burn to dedicated location in flash,
 *		use flash memory mapped address to load model.
 *		in this example, model data is pre-burn to flash address: 0x180000
```
I have a few questions about the Model location in FLASH.

1. What does `pre-burn` mean here? (we don't need to re-burn every time?)
2. Can we decide where to put the model in Flash? 
3. If we want to integrate a different model and put it in flash, how to do that?

stevehuang82, 2024-06-12:
Hi @TCHeish 

1. What does pre-burn mean here? (we don't need to re-burn every time?)
    Unless you have modified the model, you only need to burn the model once.

2. Can we decide where to put the model in Flash?
    Yes. The address of the model is defined in the following file and you can modify it yourself. The model starting address should  be 4KB aligned.
    https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/common_config.h

3. If we want to integrate a different model and put it in flash, how to do that?
    Define your model starting address in common_config.h and burn the model file in the FLASH.

TCHeish, 2024-06-12:
Hi @stevehuang82 ,

> Can we decide where to put the model in Flash?
Yes. The address of the model is defined in the following file and you can modify it yourself. The model starting address should be 4KB aligned.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/common_config.h

Thanks. How about examples like `allon_sensor_tflm_fatfs`?
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/ce498fb18327dff98f8980b38aa257bb94b533eb/EPII_CM55M_APP_S/app/scenario_app/allon_sensor_tflm_fatfs/cvapp.cpp#L169

> If we want to integrate a different model and put it in flash, how to do that?
> Define your model starting address in common_config.h and burn the model file in the FLASH.

Could you help to point out where is the burn model procedure? 
Is it inside the `we2_local_image_gen.exe` ?
(I assume there is some script/code loading model from path `model_zoo` and put it in the `FLASH_ADDR` but I can't find it.)

stevehuang82, 2024-06-12:
Hi @TCHeish 

In allon_sensor_tflm_fatfs example, we define "FLASH_XIP_MODEL=0" and convert the model file to "person_detect_model_data_vela.cc". This model is embedded in the firmware and does not need to be additionally burned into the flash memory.

If you want to define the model starting address in the FLASH, you can refer to tflm_fd_fm example. And you can burn the multiple models through python scripts as shown below.

python xmodem\xmodem_send.py --port=[your COM number] --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\tflm_fd_fm\0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo\tflm_fd_fm\1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo\tflm_fd_fm\2_il_0x32A000.tflite 0x32A000 0x00000"

example:

    python xmodem\xmodem_send.py --port=COM123 --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\tflm_fd_fm\0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo\tflm_fd_fm\1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo\tflm_fd_fm\2_il_0x32A000.tflite 0x32A000 0x00000"

You can reference by the [How to build face mesh scenario app exmple and run on WE2?](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2?tab=readme-ov-file#windows-environment-1).

TCHeish, 2024-06-12:
Hi @stevehuang82 

> In allon_sensor_tflm_fatfs example, we define "FLASH_XIP_MODEL=0" and convert the model file to "person_detect_model_data_vela.cc". This model is embedded in the firmware and does not need to be additionally burned into the flash memory.

Thanks. Just curious can we overwrite a new model to `0x180000` location?

stevehuang82, 2024-06-12:
Hi @TCHeish 

The model should be placed after address 0x200000, as the first 2MB of addresses are reserved for the firmware.
If you want to place the model at address 0x200000, you can define the variable as follows.
in

stevehuang82, 2024-06-12:
Hi @TCHeish

The model should be placed after address 0x200000, as the first 2MB of addresses are reserved for the firmware.
If you want to place the model at address 0x200000, you can define the model address variable as follows.

in ~allon_sensor_tflm_fatfs/common_config.h
#define MY_MODEL_FLASH_ADDR               (BASE_ADDR_FLASH1_R_ALIAS+0x200000)

in ~allon_sensor_tflm_fatfs/cvapp.cpp
static const tflite::Model*model = tflite::GetModel((const void *)MY_MODEL_FLASH_ADDR );

You have to burn your model file to address 0x200000 in the flash through python scripts.

python xmodem\xmodem_send.py --port=[your COM number] --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local\output_case1_sec_wlcsp\output.img --model="model_zoo\allon_sensor_tflm_fatfs\0_my_model_0x200000.tflite 0x200000 0x00000"

TCHeish, 2024-06-12:
Thanks for the explanation.

closed 2024-06-13 by stevehuang82