HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #21: App tflm_fd_fm, use GPIO

hatrinh-sj, 2024-06-15:
Hi Kris,

My requirement is simple: if a face is detected, turn on a specific GPIO pin. Is it possible to use GPIO on Vision AI V2? I use serial only for debugging. I don't need face mesh or landmark.

stevehuang82, 2024-06-17:
Hi @hatrinh-sj,

You can configure PA0(D0) as a GPIO function using the following method:

#include "hx_drv_gpio.h"

// Configure PA0(D0) as GPIO0
hx_drv_gpio_set_output(AON_GPIO0, GPIO_OUT_LOW);
hx_drv_scu_set_PA0_pinmux(SCU_PA0_PINMUX_AON_GPIO0_2, 1);

// set GPIO0 output high
hx_drv_gpio_set_out_value(AON_GPIO0, GPIO_OUT_HIGH);

// set GPIO0 output low
hx_drv_gpio_set_out_value(AON_GPIO0, GPIO_OUT_LOW);

[image]

hatrinh-sj, 2024-06-17:
Thank you. I guess I must put the command to flip the pin in function cv_fd_fm_run in cvapp_fd_fm.cpp. Is it correct?

```
    if (alg_result->num_tracked_human_targets == 0)
	{
    	//need to wait jpeg transfer done
        ...
        hx_drv_gpio_set_out_value(AON_GPIO0, GPIO_OUT_LOW);

	}
    else {
       hx_drv_gpio_set_out_value(AON_GPIO0, GPIO_OUT_HIGH);
       ...

    }
```

stevehuang82, 2024-06-18:
Yes, it is.

closed 2024-06-18 by hatrinh-sj