HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #23: Multiple GPIO pin configuration

stantonious, 2024-06-18:
Can the Grove Vision AI V2 be configured to use 8 GPIO pins (to run stepper motors)?

stevehuang82, 2024-06-19:
Hi @stantonious,

You can configure the Grove Vision AI V2 pins to GPIO function as the following.
[image]
[image]

Use the following commands to setup pin functions.

#include "hx_drv_gpio.h"

// Configure PB2(D9) as GPIO13
hx_drv_gpio_set_output(GPIO13, GPIO_OUT_LOW);
hx_drv_scu_set_PB2_pinmux(SCU_PB2_PINMUX_GPIO13, 1);

// Configure PB3(D10) as GPIO14
hx_drv_gpio_set_output(GPIO14, GPIO_OUT_LOW);
hx_drv_scu_set_PB3_pinmux(SCU_PB3_PINMUX_GPIO14, 1);

// Configure PB4(D8) as GPIO15
hx_drv_gpio_set_output(GPIO15, GPIO_OUT_LOW);
hx_drv_scu_set_PB4_pinmux(SCU_PB4_PINMUX_GPIO15, 1);

// Configure PB6(D6) as GPIO0
hx_drv_gpio_set_output(GPIO0, GPIO_OUT_LOW);
hx_drv_scu_set_PB6_pinmux(SCU_PB6_PINMUX_GPIO0_1, 1);

// Configure PB7(D7) as GPIO1
hx_drv_gpio_set_output(GPIO1, GPIO_OUT_LOW);
hx_drv_scu_set_PB7_pinmux(SCU_PB7_PINMUX_GPIO1_1, 1);

// Configure PB8(D2) as GPIO2
hx_drv_gpio_set_output(GPIO2, GPIO_OUT_LOW);
hx_drv_scu_set_PB8_pinmux(SCU_PB8_PINMUX_GPIO2_1, 1);

// Configure PA2(D5) as SB_GPIO0
hx_drv_gpio_set_output(SB_GPIO0, GPIO_OUT_LOW);
hx_drv_scu_set_PA2_pinmux(SCU_PA2_PINMUX_SB_GPIO0, 1);

// Configure PA3(D4) as SB_GPIO1
hx_drv_gpio_set_output(GPIO2, GPIO_OUT_LOW);
hx_drv_scu_set_PA3_pinmux(SCU_PA3_PINMUX_SB_GPIO1, 1);

Follow the following commands to control GPIO high/low.

// Set GPIO13 high
hx_drv_gpio_set_out_value(GPIO13, GPIO_OUT_HIGH);
// Set GPIO13 low
hx_drv_gpio_set_out_value(GPIO13, GPIO_OUT_LOW);

// Set GPIO14 high
hx_drv_gpio_set_out_value(GPIO14, GPIO_OUT_HIGH);
// Set GPIO14 low
hx_drv_gpio_set_out_value(GPIO14, GPIO_OUT_LOW);

// Set GPIO15 high
hx_drv_gpio_set_out_value(GPIO15, GPIO_OUT_HIGH);
// Set GPIO15 low
hx_drv_gpio_set_out_value(GPIO15, GPIO_OUT_LOW);

// Set GPIO0 high
hx_drv_gpio_set_out_value(GPIO0, GPIO_OUT_HIGH);
// Set GPIO0 low
hx_drv_gpio_set_out_value(GPIO0, GPIO_OUT_LOW);

// Set GPIO1 high
hx_drv_gpio_set_out_value(GPIO1, GPIO_OUT_HIGH);
// Set GPIO1 low
hx_drv_gpio_set_out_value(GPIO1, GPIO_OUT_LOW);

// Set GPIO2 high
hx_drv_gpio_set_out_value(GPIO2, GPIO_OUT_HIGH);
// Set GPIO2 low
hx_drv_gpio_set_out_value(GPIO2, GPIO_OUT_LOW);

// Set SB_GPIO0 high
hx_drv_gpio_set_out_value(SB_GPIO0, GPIO_OUT_HIGH);
// Set SB_GPIO0 low
hx_drv_gpio_set_out_value(SB_GPIO0, GPIO_OUT_LOW);

// Set SB_GPIO1 high
hx_drv_gpio_set_out_value(SB_GPIO1, GPIO_OUT_HIGH);
// Set SB_GPIO1 low
hx_drv_gpio_set_out_value(SB_GPIO1, GPIO_OUT_LOW);

stantonious, 2024-06-24:
Great, thanks!

closed 2024-06-24 by stantonious