HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #24: Using a button

cyxl, 2024-06-21:
Can we have an example of using a button with the Grove Vision AI V2

stevehuang82, 2024-06-25:
Hi @cyxl,

You can refer to [EPII_CM55M_APP_S/drivers/inc/hx_drv_gpio.h ](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/drivers/inc/hx_drv_gpio.h) to set PA0(D0) to GPIO input mode as button input.

For example, Usage-2:  Use AON_GPIO0 as an input pin with rising edge to trigger interrupt.
You can change interrupt type to rising and falling edge (GPIO_IRQ_TRIG_TYPE_EDGE_BOTH) to trigger interrupt.

In our experience, there is some bounce pulse when pressing the button, and you'll have to deal with the bounce yourself.

stantonious, 2024-06-25:
I don't see a pull-up resistor mode for input. Do I need to add a resistor and if so what size?

stevehuang82, 2024-06-25:
You can add a 10K resistor to the circuit as shown below.
[image]

stantonious, 2024-06-25:
Perfect!  Thanks again