HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #25: spi_fatfs.c incompatible with freertos?

stantonious, 2024-06-23:
Hello - We're looking to access the sd card via spi_fatfs (similar to `allon_sensor_tflm_fatfs`) while use the freertos OS.  It appears all of the sample `spi_fatfs.c` are using the eventlibrary, and are incompatible.  While porting over spi_fatfs.c, the dp_task timesout trying to acquire with the following output

```
sensor_type: 15
OV5647 on by app done
g_cursensorstream: 0
os_app_dplib_cb event=-76
send to dp task 0x11c
os_app_dplib_cb event=-77
send to dp task 0x11e
dp_recv_msg_event=0x11c
dp_recv_msg_event=0x11e
main_recv_msg=0x203
main_recv_msg.msg_data=0x11c fail
main_recv_msg=0x203
main_recv_msg.msg_data=0x11e fail
```

Is there an example spi_fatfs that uses freertos?

Thanks!

stevehuang82, 2024-06-28:
Hi @stantonious,

We don't have this example, we'll need some time to port it.

stevehuang82, 2024-07-04:
Hi @stantonious,

We've added fatfs support for the allon_sensor_tflm_freertos application.
You can enable FATFS support by setting SUPPORT_FATFS=1.

EPII_CM55M_APP_S/app/scenario_app/allon_sensor_tflm_freertos/common_config.h.
#define SUPPORT_FATFS               1       // 0 : send images via SPI, 1 : save images to SD card

stantonious, 2024-07-09:
Thankyou!