HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #26: SPI Flash Access

epibaikas, 2024-06-24:
Hi, would be possible to add an example about how the SPI Flash can be accessed? I can see there are some header files within `EPII_CM55M_APP_S/library/spi_eeprom` directory that contain functions for using the eeprom. However, when I tried to use them, I got a UsageFault_Handler interrupt.  Have the functions not been implemented yet?

stevehuang82, 2024-06-25:
Hi @epibaikas,

Do you want to read SPI flash through xip mode (read only)? Or do you want to read/write SPI flash memory? Which function did you call and get the UsageFault_Handler interrupt?

epibaikas, 2024-06-25:
Hi Steve, I want to read/write to SPI flash memory during runtime. I realised that the UsageFault_Handler was generated by a call to memcpy function that was irrelevant to the SPI flash access functions, so you can ignore that. The code I am trying to run is the following:

```
	// Initialize eeprom
	xprintf("Init EEPROM...\r\n");
	uint8_t id_info = 2;
	USE_DW_SPI_MST_E spi_id = USE_DW_SPI_MST_Q;
	hx_lib_spi_eeprom_open(spi_id);
	hx_lib_spi_eeprom_read_ID(spi_id, &id_info);

	#define NUM_OF_WORDS 2
	uint32_t tx_buf[NUM_OF_WORDS] = {0xAAAAAAAA, 0xBBBBBBBB};
	hx_lib_spi_eeprom_word_write(spi_id, 0x00000000, &tx_buf[0], 8);

	uint32_t rx_buf[NUM_OF_WORDS] = {0, 0};
	hx_lib_spi_eeprom_word_read(spi_id, 0x00000000, &rx_buf[0], 8);

	unsigned int i = 0;
	for (i = 0; i < NUM_OF_WORDS; i++) {
		printf("i = %u, rx_buf[i]=%u\r\n", i, rx_buf[0]);
	}
```

An this is the output that I get:
```
1st BL Modem Build DATE=Nov 30 2023, 0x0002000b                                                                                                                                                                    
Please input any key to enter X-Modem mode in 100 ms                                                                                                                                                               
waiting input key...                                                                                                                                                                                               
slot flash_offset 0x00100000                                                                                                                                                                                       
New MemDesp himax_sec_SB_image_process PASS                                                                                                                                                                        
set_memory_s_ns                                                                                                                                                                                                    
bl_status = 0x800000, HX_DSP_FLAG 1                                                                                                                                                                                
bl_status = 0x800000                                                                                                                                                                                               
jump_addr=0x10000000                                                                                                                                                                                               
Compiler Version: ARM GNU, 13.2.1 20231009                                                                                                                                                                         
wakeup_event=0x0,WakeupEvt1=0x0, freq=400000000                                                                                                                                                                    
__GNUC                                                                                                                                                                                                             
__mm_start_addr__ address: 3401f000                                                                                                                                                                                
Init EEPROM...                                                                                                                                                                                                     
i = 0, rx_buf[i]=0                                                                                                                                                                                                 
i = 1, rx_buf[i]=0 
```

stevehuang82, 2024-06-27:
Hi @epibaikas,

You must erase the flash contents (one sector or more) before writing to the flash memory.

	// Initialize eeprom
	xprintf("Init EEPROM...\r\n");
	uint8_t id_info = 2;
	USE_DW_SPI_MST_E spi_id = USE_DW_SPI_MST_Q;
	hx_lib_spi_eeprom_open(spi_id);
	hx_lib_spi_eeprom_read_ID(spi_id, &id_info);
	printf("flash id = 0x%x\r\n", id_info);

	#define FLASH_ADDRESS 0x00300000
	hx_lib_spi_eeprom_erase_sector(spi_id, FLASH_ADDRESS, FLASH_SECTOR);

	#define NUM_OF_WORDS 2
	uint32_t tx_buf[NUM_OF_WORDS] = {0xAAAAAAAA, 0xBBBBBBBB};
	
	hx_lib_spi_eeprom_word_write(spi_id, FLASH_ADDRESS, &tx_buf[0], 8);

	uint32_t rx_buf[NUM_OF_WORDS] = {0, 0};
	hx_lib_spi_eeprom_word_read(spi_id, FLASH_ADDRESS, &rx_buf[0], 8);

	unsigned int i = 0;
	for (i = 0; i < NUM_OF_WORDS; i++) {
		printf("i = %d, rx_buf[i]=%x\r\n", i, rx_buf[i]);
	}

And you will get the correct output.
     Init EEPROM...
     flash id = 0xef
     i = 0, rx_buf[i]=aaaaaaaa
     i = 1, rx_buf[i]=bbbbbbbb

Note that you cannot write to the first 2MB of addresses reserved for firmware. If the bootloader is overwritten, the board will be damaged.

epibaikas, 2024-06-27:
Ok many thanks @stevehuang82! Is it safe to say that all the flash addresses >= 0x200000 are free to use assuming that we have not allocated any of them for storing the parameters of a tensorflow-lite model?

stevehuang82, 2024-06-28:
Yes, except for the last sector of flash memory (16MB). It is used for booting index.

epibaikas, 2024-06-28:
Ok, thank you for all the info. It could be useful to add an address map for the flash memory showing the regions reserved for firmware and boot index as well as the regions that are free for the users to use.

closed 2024-06-28 by epibaikas