HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #28: jlink  vs debug error

duhengxuan, 2024-06-25:
Hi, as follwed /swddebugging README.md ,when start debugging  ,it have the error :

=false -Oreset_type='sw_sysresetreq' -Oreset.hold_time=0.1 -Oreset.post_delay=1 -Oreset.halt_timeout=2 -Ostep_into_interrupt=false -Ovector_catch='h' -Oenable_semihosting=false -Osmart_flash=false -Oserve_local_only=true -Odap_swj_enable=true -Odap_swj_use_dormant=false -Odap_protocol=swd
0004042 I Target type is we2 [board]
0005439 C Memory transfer fault (Unspecified error.) [__main__]
Traceback (most recent call last):
  File "/home/hengxuan/.local/lib/python3.8/site-packages/pyocd/probe/jlink_probe.py", line 293, in read_dp
    value = self._link.coresight_read(addr // 4, ap=False)
  File "/home/hengxuan/.local/lib/python3.8/site-packages/pylink/jlink.py", line 204, in wrapper
    return func(self, *args, **kwargs)
  File "/home/hengxuan/.local/lib/python3.8/site-packages/pylink/jlink.py", line 3388, in coresight_read
    raise errors.JLinkException(res)
pylink.errors.JLinkException: Unspecified error.

ps: JLINK_7.6.8c

flagyan, 2024-06-28:
Hi @duhengxuan,

Do you have any progress on this issue? I also met this problem during the debug setup. Thanks.

stevehuang82, 2024-06-28:
Hi @duhengxuan,

When using the swd debugger, the hardware requirement is a CMSIS-DAP compliant debugger. We tried connecting to the Grove Vision AI V2 using the JLink debugger, but it didn't work.
We found some information on how to configure J-Link to support CMSIS-DAP. Maybe you can try it.

https://www.segger.com/news/segger-j-link-now-supports-cmsis-dap/

https://forum.segger.com/index.php/Thread/7965-SOLVED-CMSIS-DAP/#:~:text=With%20the%20latest%20version%20of%20the%20Software%20and,Select%20%22Configure%22%20%3E%20Select%20%22CMSIS-DAP%22%20as%20boot%20mode

closed 2024-07-22 by stevehuang82