HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #30: how to download the firmware image larger than 1MB

flagyan, 2024-06-26:
Hi,

Is there a method to download firmware image larger than 1MB? I have built my library in the firmware and got a 1.5MB .img file. Thanks in advance.

stevehuang82, 2024-06-27:
Hi @flagyan,

Firmware image size cannot be larger than 1MB. Does your code include model files? If so, you can put the model file into flash memory.

flagyan, 2024-06-27:
Hi @stevehuang82,

I have a very big library to implement complex functions and so result in a 1.5M image file. Where is the limitation of 1MB from? Is  it from the download function? Then can I use the debugger to flash it directly? Thanks.

stevehuang82, 2024-06-27:
Yes, the limitation is from the download function. I think you can use the debugger to flash it directly. I will confirm that later.

flagyan, 2024-06-28:
Hi @stevehuang82 ,

I am setting up the debug environment with jlink and get the 'Unspecified error' as #28,  do we need any extra setting besides
[https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/swd_debugging/README.md](swd_debugging/README.md)? for example, enable the SWD in the image, I see that in the [https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/allon_sensor_tflm_freertos/allon_sensor_tflm.c](allon_sensor_tflm.c), there is code to enable the swd but it is commented out. Thanks a lot.

flagyan, 2024-07-01:
Hi @stevehuang82,

I can download the image with a cmsis-dap debugger now. Thanks a lot.

closed 2024-07-02 by flagyan