HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #33: Old version of tflite?

stantonious, 2024-06-29:
I'm trying to work through the example of deploying a custom YOLO model to the Grove AI V2.  During tensor allocation, i get the following:

Didn't find op for builtin opcode 'PADV2' version '2'. An older version of this builtin might be supported. Are you using an old TFLite binary with a newer model?

I'm using the COCO_Detetection_Swift notebook from here (https://wiki.seeedstudio.com/ModelAssistant_Introduce_Quick_Start/#model-training) which appears to want tensorflow>=2.12.0

stantonious, 2024-06-30:
Also - Can you provide an example `vela` command to generate a vela tflite for ethos optimizations?

THanks!

kris-himax, 2024-07-01:
Hi @stantonious ,
You should add all the ops which are at the tflite. 
[image]

You can reference the code at [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_pose/cvapp_yolov8_pose.cpp#L317).

You can find the OP ,AddPadV2() , at [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/library/inference/tflmtag2209_u55tag2205/tensorflow/lite/micro/micro_mutable_op_resolver.h#L415).

But you also need to add the OPs like reshape, gather and transpose to op resolver at your code.

And about the `vela` command, you can reference [this part](https://github.com/HimaxWiseEyePlus/ML_FVP_EVALUATION?tab=readme-ov-file#how-to-use-himax-config-file-to-generate-vela-model).
```
vela --accelerator-config ethos-u55-64 --config himax_vela.ini --system-config My_Sys_Cfg --memory-mode My_Mem_Mode_Parent --output-dir ./img_person_detect ./img_person_detect/person_int8_model.tflite
```
Thanks,
Kris

closed 2024-07-09 by kris-himax