HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #35: Read/Write latency of the memory

flagyan, 2024-07-05:
Hi,

The we2 has itcm/dtcm, sram0/1/2 and flash, what is the read/write latency of those memory? Can the sram0/1/2 be read/written in 1 clock at 400MHz (zero-wait)? Do you have this latency data for the external flash? Thanks a lot.

stevehuang82, 2024-07-22:
Hi @flagyan,

Memory read/write latency depends on system configuration.
What concerns do you have about memory read/write latency in your system?

flagyan, 2024-07-24:
Hi @stevehuang82,

I have an audio algorithm running on the M55 core and I want to know the real MIPS of this algorithm. As all the code/data are in the sram and so I would like to know the latency of those memory. Thanks a lot.

stevehuang82, 2024-07-31:
Hi @flagyan,

You can use the SystemGetTick() function to measure the running time of your audio algorithm in clock cycles. Below is an example for your reference.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_pose/cvapp_yolov8_pose.cpp#L763

flagyan, 2024-08-02:
Hi @stevehuang82 ,

Currently I read the dwt register directly for the MCPS measurement. For the memory latency, what I want is to place the most critical code to the fastest memory if the read/write latency of the ram is not the same. Thanks.

stevehuang82, 2024-08-02:
Hi @flagyan,

The itcm/dtcm read/write latency is 1 cycle/32bits.
The latency sequcnce is :  itcm/dtcm < sram 0/1 < sram 2 << flash. It is recommended to place the most critical code in itcm/dtcm.

closed 2024-08-21 by stevehuang82