HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #37: swift_yolo coco model outputs.

stantonious, 2024-07-15:
I'm able to train a custom model using sscma following instructions here https://wiki.seeedstudio.com/grove_vision_ai_v2_sscma/.  And the model run rather well using https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process!  I am able to deploy the model and my code with xmodem, run my model and perform inference.  I need to understand how to preprocess the model inputs (i'm currently using the example in tflm_yolov8_od) and post process the model outputs.  It appears I'm pretty close (my model has 8 classes and I get a 1,n,13 output where 13 should be xmin,ymin,width,height,obj,class1....class8).  I can't make sense of the bounding box provided and the confidences are > 1 (like between 1 and 100).  

Can you please provide details on quantization of both the model inputs and outputs or provide an example of the code that SenseCraft uses to deploy the model?  

Thanks!

Model used configs/swift_yolo/swift_yolo_shuff_1xb16_300e_coco.py

kris-himax, 2024-07-16:
Hi @stantonious ,

Maybe you should reference the github repository at https://github.com/Seeed-Studio/sscma-example-we2/tree/main

The repository uses the SSCMA library and algorithm from Seeed.
https://github.com/Seeed-Studio/SSCMA-Micro/tree/c57630182508a775b4f78d95ebe57bc13180a171/core/algorithm

Thanks,
Kris

stantonious, 2024-07-17:
Worked perfectly!  Thanks

closed 2024-07-17 by stantonious