HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #42: face mesh scenario_app not working

103061634, 2024-08-07:
Hi,
We used the Seeed Grove Vision AI Module V2 and followed your instructions to set up the face mesh scenario_app project, but it seems there is a problem with camera streaming. Here is the log:

`
b'Face Detection Face mesh'
b'Ethos-U55 device initialised'
b"FD model's schema version 3"
b"FM model's schema version 3"
b"IL model's schema version 3"
b'initial done'
b'cis_OV5647_init'
b'mclk DIV3, xshutdown_pin=82'
b'Set PA1(AON_GPIO1) to High'
b'hx_drv_cis_set_slaveID(0x36)'
b'dw_iic_write err_state:2'
b'dw_iic_write err_code:-60'
b'[hx_drv_i2cm_write_data]-L:701, Fail:-60'
b'retry=0, (regAddr=0x4800, val=0x01) Fail (-60)'
b'dw_iic_write err_state:2'
b'dw_iic_write err_code:-60'
b'[hx_drv_i2cm_write_data]-L:701, Fail:-60'
b'retry=1, (regAddr=0x4800, val=0x01) Fail (-60)'
b'dw_iic_write err_state:2'
b'dw_iic_write err_code:-60'
b'[hx_drv_i2cm_write_data]-L:701, Fail:-60'
b'retry=2, (regAddr=0x4800, val=0x01) Fail (-60)'
b'OV5647 off by app fail'
b''
b'CIS Init fail'
b'read_data=2999994 not reset'
b'CLI_WDG_Reset_ISR_CB event=1'
`
Thanks!!

kris-himax, 2024-08-07:
Hi @103061634 

Please make sure your camera is plugged at the right direction. You can reference following video.
https://wiki.seeedstudio.com/grove_vision_ai_v2/#connecting-to-a-csi-interface-camera

kris-himax, 2024-08-08:
@103061634 

Or make sure the camera connection in the picture below is firmly pressed.
[image]

103061634, 2024-08-08:
Hi Kris,
Thank you for your response. After checking, we found that the issue was with the camera, which is now functioning normally. 
On the other hand, I would like to ask how to convert ticks into milliseconds. After tracing the code, the conversion may be "tick / (SystemCoreClock / 1000)" by "himax-we2/ei_classifier_porting.cpp". Is this correct?

kris-himax, 2024-08-08:
@103061634 ,
You can reference here.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cvapp_yolov8n_ob.cpp#L603

the result of following code should be convert to second
```
uint32_t  result_tick = (loop_cnt_2-loop_cnt_1)*CPU_CLK+(systick_1-systick_2)
```
then let 

The unit of this value (result_tick / 400000000) is second.

103061634, 2024-08-08:
Hi Kris,
thanks a lot!

closed 2024-08-08 by 103061634