HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #48: Raw Image Not Updating

QuanTrinhCA, 2024-08-21:
Hello,

I am trying to capture raw images from the camera (OV5647) but I am facing issues with the image data not updating. Reseting the board allows the image data to be updated but using sensordplib_retrigger_capture() does not. With sensordplib_retrigger_capture(), the event XDMA_FRAME_READY is triggered but the image data remains the same as the initial capture. Could I be advised on a potential solution to this?

Here is a snippet of the code:
```
void evt_handler_db_cb(EVT_INDEX_E event) {
    switch (event) {
        case EVT_INDEX_XDMA_FRAME_READY:
            xprintf("Frame ready\n");
            //Test print image data
            uint8_t *img = app_get_raw_addr();
            uint32_t width = app_get_raw_width();
            uint32_t height = app_get_raw_height();
            xprintf("Image addr: %d | width: %d | height: %d\n", img, width, height);
            for (int h = 0; h < height/50; h++) {
                for (int w = 0; w < width/50; w++)
                {
                    xprintf("%d ", *img++);
                }
                xprintf("\n");
            }
            for (uint64_t i = 0; i < 999999999; i++) {
                __NOP();
            }
            sensordplib_retrigger_capture();
            break;
```

Thank you.

stevehuang82, 2024-08-21:
Hi @QuanTrinhCA,

Your problem should be cache issue. Because the sensor image is transferred to SRAM by DMA. If you want to access the sensor image by CPU, you should invalid the DCache.
Please modify your code as the following.

//Test print image data
uint8_t *img = app_get_raw_addr();
### uint32_t img_size = app_get_raw_sz();
### SCB_InvalidateDCache_by_Addr((void *)img, img_size);
uint32_t width = app_get_raw_width();
uint32_t height = app_get_raw_height();
xprintf("Image addr: %d | width: %d | height: %d\n", img, width, height);

Steve

QuanTrinhCA, 2024-08-21:
Thank you very much, that was the issue.

closed 2024-08-21 by QuanTrinhCA