HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #49: Failed to get registration from op code CUSTOM

QuanTrinhCA, 2024-08-22:
Hello,

I am trying to use [Ultra-Light-Fast-Generic-Face-Detector-1MB](https://github.com/Linzaer/Ultra-Light-Fast-Generic-Face-Detector-1MB/tree/master) model in my app. 
I have converted the onnx model to tflite 8int using onnx2tf:
`onnx2tf -i version-RFB-320_without_postprocessing.onnx -cind "input" 'data/calibdata.npy' "[[[[0.46597886, 0.42701387, 0.41232178]]]]" "[[[[0.30210555, 0.29369238, 0.29560316]]]]" -b 1 -oiqt -osd -ioqd int8 -cotof`
Then, I used vela to optimize the model:
`vela --accelerator-config ethos-u55-64 --config himax_vela.ini --system-config My_Sys_Cfg --memory-mode My_Mem_Mode_Parent --output-dir ./img_person_detect ML_FVP_EVALUATION/vela/saved_model/version-RFB-320_without_postprocessing_full_integer_quant.tflite`
Which nets this ouput:
```
Network summary for version-RFB-320_without_postprocessing_full_integer_quant
Accelerator configuration                Ethos_U55_64
System configuration                       My_Sys_Cfg
Memory mode                        My_Mem_Mode_Parent
Accelerator clock                                 400 MHz
Design peak SRAM bandwidth                       3.20 GB/s
Design peak Off-chip Flash bandwidth             0.05 GB/s

Total SRAM used                               1518.75 KiB
Total Off-chip Flash used                      352.02 KiB

CPU operators = 0 (0.0%)
NPU operators = 95 (100.0%)

Average SRAM bandwidth                           1.32 GB/s
Input   SRAM bandwidth                          17.19 MB/batch
Weight  SRAM bandwidth                           3.54 MB/batch
Output  SRAM bandwidth                          11.33 MB/batch
Total   SRAM bandwidth                          32.11 MB/batch
Total   SRAM bandwidth            per input     32.11 MB/inference (batch size 1)

Average Off-chip Flash bandwidth                 0.01 GB/s
Input   Off-chip Flash bandwidth                 0.00 MB/batch
Weight  Off-chip Flash bandwidth                 0.31 MB/batch
Output  Off-chip Flash bandwidth                 0.00 MB/batch
Total   Off-chip Flash bandwidth                 0.31 MB/batch
Total   Off-chip Flash bandwidth  per input      0.31 MB/inference (batch size 1)

Neural network macs                         147214760 MACs/batch
Network Tops/s                                   0.01 Tops/s

NPU cycles                                    9228022 cycles/batch
SRAM Access cycles                            3704408 cycles/batch
DRAM Access cycles                                  0 cycles/batch
On-chip Flash Access cycles                         0 cycles/batch
Off-chip Flash Access cycles                     7680 cycles/batch
Total cycles                                  9727349 cycles/batch

Batch Inference time                24.32 ms,   41.12 inferences/s (batch size 1)
```
I then uploaded the model to flash and was able to read the model but have the error:
```
b'Model is schema version 3'
b'Failed to get registration from op code CUSTOM'
b''
b'TFLM initialised'
```
I checked the model and it does not look like there are any custom ops:
[image]
[image]
In the code, I have already added ethosu op:
```
int tflite_init() {
    //error_reporter = tflite::MicroErrorReporter();

    tensor_arena = (uint8_t *)mm_reserve(TENSOR_ARENA_SIZE);

    static const tflite::Model *model = tflite::GetModel((const void *)(BASE_ADDR_FLASH1_R_ALIAS+0x200000));
   
    if (model->version() != TFLITE_SCHEMA_VERSION)
    {
        xprintf("Model provided is schema version %d, but required version is %d\n",
                model->version(), TFLITE_SCHEMA_VERSION);
        return -1;
    } else {
        xprintf("Model is schema version %d\n", model->version());
    }
    
    if (kTfLiteOk != op_resolver.AddEthosU())
    {
        xprintf("Failed to add EthosU op resolver\n");
        return -1;
    }

    static tflite::MicroInterpreter interpreter(model, 
                                                op_resolver,
                                                tensor_arena,
                                                TENSOR_ARENA_SIZE);
    interpreter.AllocateTensors();
    interpreter_ptr = &interpreter;
    xprintf("TFLM initialised\n");
    return 0;
}
```
I am unsure where this error could arise from. Could I be advised on a potential cause of this error?

kris-himax, 2024-08-22:
Hi @QuanTrinhCA ,

I am not sure about how you arrange the memory. 
But I see your vela report about  `Total SRAM used                               1518.75 KiB` , I think that it is too high (WE2 internal SRAM is only 2MB). Because at your SRAM you may not only put tensor_arena (SRAM) but also sensor image and other parameter.
You can try to add following command at your vela command to limit the SRAM size to 1MB or 900KB by using `6.` of https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ethos-u-vela/#running.
```
--optimise Performance --arena-cache-size 1000000
```

QuanTrinhCA, 2024-08-22:
Hello @kris-himax ,

Thanks for the response. I have tried the command but I still get the error. I have also tried using older versions of the conversion tools to correspond to TensorFlow 2.16.1 but it still gives the error. Upon closer inspection, it seems that the default model in this github also get that error. Thus, I changed the TFLM library from 2405 to default 2209 and the default model seems to work again. It seems that 2405 gives:
```
b'Name: \xb5\x02H\x14\xf0t\xf8\xfe\xe7 Code: 268443317'
b'Failed to get registration from op code CUSTOM'
```
with this test code:
```
TFLMRegistration* = tflite::Register_ETHOSU();
xprintf("Name: %s", registration->custom_name);
//registration->builtin_code=0;
xprintf(" Code: %d\n", registration->builtin_code);
```
While 2209 gives:
```
b'Name:  Code: 0'
```
without any errors with this test code:
```
TfLiteRegistration* = tflite::Register_ETHOSU();
xprintf("Name: %s", registration->custom_name);
//registration->builtin_code=0;
xprintf(" Code: %d\n", registration->builtin_code);
```
The output of the register function is not the same between the two versions. It looks like `tflite::Register_ETHOSU()` internally calls the first of these two functions:
```
TFLMRegistration RegisterOp(
    void* (*init)(TfLiteContext* context, const char* buffer, size_t length),
    TfLiteStatus (*prepare)(TfLiteContext* context, TfLiteNode* node),
    TfLiteStatus (*invoke)(TfLiteContext* context, TfLiteNode* node),
    void (*free)(TfLiteContext* context, void* buffer),
    void (*reset)(TfLiteContext* context, void* buffer)) {
  return {/*init=*/init,
          /*free=*/free,
          /*prepare=*/prepare,
          /*invoke=*/invoke,
          /*reset*/ reset,
          /*builtin_code=*/0,
          /*custom_name=*/nullptr};
}

TFLMInferenceRegistration RegisterOp(
    TfLiteStatus (*invoke)(TfLiteContext* context, TfLiteNode* node),
    void (*reset)(TfLiteContext* context, void* buffer)) {
  return {
      /*invoke=*/invoke,
      /*reset*/ reset,
  };
}
```
```
TFLMRegistration* Register_ETHOSU() {
  static TFLMRegistration r = tflite::micro::RegisterOp(Init, Prepare, Eval);
  return &r;
}
```
which should have returned nullptr for name and 0 for op code but it does not. I am unsure as to why this is the case. Do you have any insights regarding this?

I have downloaded release 24.05 from ARM and was able to compile successfully with this makefile:
```
# directory declaration
LIB_INFERENCE_ENGINE_DIR = $(LIBRARIES_ROOT)/inference/tflmtag2405_u55tag2405
LIB_INFERENCE_ENGINE_DEFINES = -DTFLM2405_U55TAG2405

# use for information only rules.mk - ALL_ASMSRC_DIRS
LIB_INFERENCE_ENGINE_ASMSRCDIR	= $(LIB_INFERENCE_ENGINE_DIR)
LIB_INFERENCE_ENGINE_CSRCDIR	= $(LIB_INFERENCE_ENGINE_DIR) 
LIB_INFERENCE_ENGINE_CXXSRCSDIR    = $(LIB_INFERENCE_ENGINE_DIR)

[... 401 lines cut ...]
INFERENCE_ENGINE_LIB_CXXOBJS += $(LIB_INFERENCE_ENGINE_CXXOBJS)
INFERENCE_ENGINE_LIB_ASMOBJS += $(LIB_INFERENCE_ENGINE_ASMOBJS)
INFERENCE_ENGINE_LIB_ALLOBJS += $(LIB_INFERENCE_ENGINE_OBJS)

LIB_DEFINES += $(LIB_INFERENCE_ENGINE_DEFINES)
LIB_DEPS += $(LIB_INFERENCE_ENGINE_DEPS)
# INFERENCE_ENGINE_LIB_LIBS += $(LIB_LIB_INFERENCE_ENGINE)

# library generation rule
ifeq ($(INFERENCE_FORCE_PREBUILT), y)
override LIB_INFERENCE_ENGINE_OBJS:=
endif

INFERENCE_ENGINE_LIB := $(subst /,$(PS), $(strip $(OUT_DIR)/$(INFERENCE_ENGINE_LIB_NAME)))

# library generation rule
$(INFERENCE_ENGINE_LIB): $(LIB_INFERENCE_ENGINE_OBJS)
	$(TRACE_ARCHIVE)
ifeq "$(strip $(LIB_INFERENCE_ENGINE_OBJS))" ""
	$(CP) $(PREBUILT_LIB)$(INFERENCE_ENGINE_LIB_NAME) $(INFERENCE_ENGINE_LIB)
else
	$(Q)$(AR) $(AR_OPT) $@ $(LIB_INFERENCE_ENGINE_OBJS)
	$(CP) $(INFERENCE_ENGINE_LIB) $(PREBUILT_LIB)$(INFERENCE_ENGINE_LIB_NAME)
endif
```

kris-himax, 2024-08-23:
Hi @QuanTrinhCA 

> `I have downloaded release 24.05 from ARM`.

Did you mean that you download the `ethos-u-core-driver` about tag 24.05?
And which GitHub commit version about TFLM did you download?
In my opinion, If you want to use latest version about U55 driver, you should download the closer version about TFLM.

And you should also change the cache function at ethosu_driver.c with HIMAX SDK yourself.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/library/inference/tflmtag2209_u55tag2205/third_party/ethos_u_core_driver/src/ethosu_driver.c#L133
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/library/inference/tflmtag2209_u55tag2205/third_party/ethos_u_core_driver/src/ethosu_driver.c#L144

Thanks,
Kris

QuanTrinhCA, 2024-08-23:
Hello @kris-himax ,

I have used the `fetch_externals.py` from [here](https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ethos-u/+/refs/tags/24.05) with `24.05.json`:
```
{
    "externals":
    [
        {
            "path": "core_platform",
            "fetchurl": "https://review.mlplatform.org/ml/ethos-u/ethos-u-core-platform",
            "pushurl": "ssh://review.mlplatform.org:29418/ml/ethos-u/ethos-u-core-platform",
            "revision": "24.05"
        },
        {
            "path": "core_software",
            "fetchurl": "https://review.mlplatform.org/ml/ethos-u/ethos-u-core-software",
            "pushurl": "ssh://review.mlplatform.org:29418/ml/ethos-u/ethos-u-core-software",
            "revision": "24.05"
        },
        {
            "path": "core_software/core_driver",
            "fetchurl": "https://review.mlplatform.org/ml/ethos-u/ethos-u-core-driver",
            "pushurl": "ssh://review.mlplatform.org:29418/ml/ethos-u/ethos-u-core-driver",
            "revision": "24.05"
        },
        {
            "path": "core_software/cmsis",
            "fetchurl": "https://github.com/ARM-software/CMSIS_5.git",
            "revision": "51f8f89612c8330c0116479cfe3a2cb04c390e2b"
        },
        {
            "path": "core_software/cmsis_6",
            "fetchurl": "https://github.com/ARM-software/CMSIS_6.git",
            "revision": "v6.1.0"
        },
        {
            "path": "core_software/cmsis_6/Cortex_DFP",
            "fetchurl": "https://github.com/ARM-software/Cortex_DFP.git",
            "revision": "v1.1.0"
        },
        {
            "path": "core_software/cmsis-nn",
            "fetchurl": "https://github.com/ARM-software/CMSIS-NN.git",
            "revision": "v6.0.0"
        },
        {
            "path": "core_software/cmsis-view",
            "fetchurl": "https://github.com/ARM-software/CMSIS-View.git",
            "revision": "pack/1.2.0"
        },
        {
            "path": "core_software/openamp/libmetal",
            "fetchurl": "https://github.com/OpenAMP/libmetal",
            "revision": "959fddbd706a67b0699e0dd7a46e36fe49079ef9"
        },
        {
            "path": "core_software/openamp/openamp",
            "fetchurl": "https://github.com/OpenAMP/open-amp",
            "revision": "1680aada793d78b512f98cc1fa8fc2ee837919e5"
        },
        {
            "path": "core_software/rtos/freertos/freertos_kernel",
            "fetchurl": "https://github.com/FreeRTOS/FreeRTOS-Kernel.git",
            "revision": "V11.0.1"
        },
        {
            "path": "core_software/rtos/threadx",
            "fetchurl": "https://github.com/eclipse-threadx/threadx",
            "revision": "v6.4.1_rel"
        },
        {
            "path": "core_software/tflite_micro",
            "fetchurl": "https://github.com/tensorflow/tflite-micro",
            "revision": "8e22946b3faa51564df5dd9194f7540b2694892c"
        },
        {
            "path": "linux_driver_stack",
            "fetchurl": "https://review.mlplatform.org/ml/ethos-u/ethos-u-linux-driver-stack",
            "pushurl": "ssh://review.mlplatform.org:29418/ml/ethos-u/ethos-u-linux-driver-stack",
            "revision": "24.05"
        },
        {
            "path": "vela",
            "fetchurl": "https://review.mlplatform.org/ml/ethos-u/ethos-u-vela",
            "pushurl": "ssh://review.mlplatform.org:29418/ml/ethos-u/ethos-u-vela",
            "revision": "3.12.0"
        }
    ]
}
```
 The result is this:
[image]
I then copied the tensorflow lib and u55 driver from `core_software` and replicate it as close as possible to the default library.
[image]
I did not know about the cache functions so I will try that next.

kris-himax, 2024-08-26:
Hi @QuanTrinhCA ,

I think that you can use the FVP which is the simulate platform at PC to test your model first.
And you can modify the downloading version about the driver and tflm version 
 at [here](https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ml-embedded-evaluation-kit/+/refs/tags/24.05/scripts/py/dependency_urls.json).
https://review.mlplatform.org/plugins/gitiles/ml/ethos-u/ml-embedded-evaluation-kit/

QuanTrinhCA, 2024-08-26:
@kris-himax Thank you for the response, I will try and report back after I settled down for the term.

closed 2024-09-26 by kris-himax