HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #51: Weird put

benclockworks, 2024-08-31:
Hello I have s**uccessfully built and flashed the ei_standalone_inferencing_camera application** [https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2.git ] **but I keep getting this as outputs**:
�pF0FPp_�?6Vs�o8B�4vx��4�� V�SXvtsV�N�0_8�xp�����

To:
6rC4zF�21tKs��6Vs68B��^3�4V�
                            R�S4szPpt�^�VT��V��W>�V
or this when I use PlatformIO's serial monitor tool:R;p4:vP×802t0ó?7vÿ4x4ø?ø8UßCÛSXFV3VO8C~ÿ<8ÿO:Px0
**ANY USEFUL HELP WILL BE APPRECIATED even links to non Arduino IDE guides will be more helpful to me than Arduino IDE guides.**
**Steps I used for flashing** :I press reset and press boot and then flash and after I flash it I use screen /minicom  
any help will be very helpful  
**Hardware**: Grove vision Ian module , FPC , and raspberry pi camera 
My steps above might be wrong or the fact that the puny code deprication warning might be stopping me,
proof of successful build and flash:# Add other commands for building your project here
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % himax-flash-tool -d WiseEye2 -f obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCS
(node:24383) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instea
(Use `node --trace-deprecation ...` to show where the warning was created)
[HMX] Connecting to /dev/tty.usbmodem578D0274271...
[HMX] Connected, press the **RESET** button on your Himax Grove Vision AI Module V2 now
[HMX] Restarted into bootloader. Sending file.
[HMX] Sending 5123 blocks
 ████████████████████████████████████████ 100% | ETA: 0s | 5123/5123^C
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % himax-flash-tool -d WiseEye2 -f obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCS
(node:26617) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instea
(Use `node --trace-deprecation ...` to show where the warning was created)
[HMX] Connecting to /dev/tty.usbmodem578D0274271...
[HMX] Connected, press the **RESET** button on your Himax Grove Vision AI Module V2 now
[HMX] Restarted into bootloader. Sending file.
[HMX] Sending 5123 blocks
 ████████████████████████████████████████ 100screen /dev/tty.usbmodem578D0274271 115200                                
[screen is terminating]
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % himax-flash-tool -d WiseEye2 -f obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCS
(node:27263) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instea
(Use `node --trace-deprecation ...` to show where the warning was created)
[HMX] Connecting to /dev/tty.usbmodem578D0274271...
[HMX] Connected, press the **RESET** button on your Himax Grove Vision AI Module V2 now
[HMX] Restarted into bootloader. Sending filemake all                                                                  
app/scenario_app/ei_standalone_inferencing_camera/ei_standalone_inferencing_camera.mk:30: invalid `override' directive
app/scenario_app/ei_standalone_inferencing_camera/ei_standalone_inferencing_camera.mk:48: Extraneous text after `ifeq' d
Compiling             :  app/scenario_app/event_handler/event_handler.c
arm-none-eabi-gcc -mthumb -mcpu=cortex-m55  -DARMCM55 -mfloat-abi=hard -c -ffunction-sections -fdata-sections -Wall   -f
Linking               :  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf
arm-none-eabi-gcc -DARM_MATH_DSP -DARM_MATH_LOOPUNROLL -DARM_MATH_MVEI -DCM55_BIG -DCOREV_0P9V -DDEBUG -DEIDSP_LOAD_CMSI
arm-none-eabi-gcc: error: objs.in: No such file or directory
make: *** [obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf] Error 1
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % touch objs.in
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % make all     
app/scenario_app/ei_standalone_inferencing_camera/ei_standalone_inferencing_camera.mk:30: invalid `override' directive
app/scenario_app/ei_standalone_inferencing_camera/ei_standalone_inferencing_camera.mk:48: Extraneous text after `ifeq' d
Linking               :  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf
arm-none-eabi-gcc -DARM_MATH_DSP -DARM_MATH_LOOPUNROLL -DARM_MATH_MVEI -DCM55_BIG -DCOREV_0P9V -DDEBUG -DEIDSP_LOAD_CMSI
/Users/benchikbayor/esp/Seeed_Grove_Vision_AI_Module_V2/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3
Memory region         Used Size  Region Size  %age Used
 CM55M_S_APP_ROM:          20 B       256 KB      0.01%
CM55M_S_APP_DATA:        1112 B       256 KB      0.42%
    CM55M_S_SRAM:         324 B      1924 KB      0.02%
arm-none-eabi-size obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf
   text	   data	    bss	    dec	    hex	filename
    324	   1100	     32	   1456	    5b0	obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.
rm -rf objs.in
objs.in created
# Add other commands for building your project here
benchikbayor@benchiks-Mac EPII_CM55M_APP_S % himax-flash-tool -d WiseEye2 -f obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCS
(node:27375) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instea
(Use `node --trace-deprecation ...` to show where the warning was created)
[HMX] Connecting to /dev/tty.usbmodem578D0274271...
[HMX] Connected, press the **RESET** button on your Himax Grove Vision AI Module V2 now
[HMX] Restarted into bootloader. Sending filescreen /dev/tty.usbmodem578D0274271 115200                                
[screen is terminating]

benclockworks, 2024-08-31:
By the way I do have to manually touch obis.in since the make all command won't work without it is that normal>

kris-himax, 2024-09-02:
@benclockworks 

You should reference the readme about [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/ei_standalone_inferencing_camera#how-to-flash-your-firmware).
You should flash the binary image file by the command.
```
himax-flash-tool -d WiseEye2 -f output_case1_sec_wlcsp/output.img
```

kris-himax, 2024-09-02:
Or you can try to flash the binary image file by Python code, it will be more easier.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2?tab=readme-ov-file#flash-image-update-at-linux-environment-by-python-code

benclockworks, 2024-09-02:
Thank you for the more specific instructions:I got this output now which better than the last one for APP_TYPE = ei_standalone_inferencing_camera :l/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img] done
exit status code: 0
signed (or raw sec type) file name = sign_formal_cm55m_s_application_p2.img
AddFlashAlignPadding
/Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img
Output image: output_case1_sec_wlcsp/cm55m_s_application.img
There is no cfg or model section

SEC_MEM_PREPROCESSING
start_addr: 0x00000000, end_addr: 0x0003ffff
start_addr: 0x20000000, end_addr: 0x2003ffff
start_addr: 0x24000000, end_addr: 0x240fffff
start_addr: 0x24100000, end_addr: 0x241fffff
start_addr: 0x26000000, end_addr: 0x2605ffff
start_addr: 0x2a000000, end_addr: 0x2affffff

SEC MEM GEN DONE...

SEC_MEM_PROCESSING

SEC_MEM_PREPROCESSING_OTA
start_addr: 0x00000000, end_addr: 0x0003ffff
start_addr: 0x20000000, end_addr: 0x2003ffff
start_addr: 0x24000000, end_addr: 0x240fffff
start_addr: 0x24100000, end_addr: 0x241fffff
start_addr: 0x26000000, end_addr: 0x2605ffff
start_addr: 0x2a000000, end_addr: 0x2affffff

SEC MEM GEN DONE...

SEC_MEM_PROCESSING_OTA

HX_MEM_DESCRIPTOR_PROCESSING
Support: bootloader- part_1
Support: 2nd_bootloader- part_2
Support: hx_memory_descriptor- part_3
Support: hx_memory_descriptor_ota- part_4
Support: cm55m_s_application- part_5
Support: cm55m_s_application- part_6
['part_0'] mem_list: [81920, 163839] (bootloader)
['part_1'] mem_list: [4096, 81919] (2nd_bootloader)
['part_2'] mem_list: [0, 4095] (hx_memory_descriptor)
['part_3'] mem_list: [163840, 167935] (hx_memory_descriptor_ota)
['part_4'] mem_list: [167936, 172031] (cm55m_s_application)
['part_5'] mem_list: [172032, 176127] (cm55m_s_application)
header
part_1
part_2
part_3
part_4
part_5
part_6
***************** Add secureboot certificate header *****************
create_image_tbl file_name =/Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/output_case1_sec_wlcsp/hx_layout.bin
create_image_tbl sram_addr =0x34000000, flash_addr=0x3a0004ac, enc_flag=0
sec_attr scheme = ICV
Hx key cert: /Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/secureboot_tool/cert/ICVSBKey1.crt
input file : Images.txt
/Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/output_case1_sec_wlcsp/sign_formal_hx_layout.bin
input base name : Images.txt
sec_format = BLP
2024-09-02 08:48:31,461 - Content Certificate Generation Utility started (Logging to sb_content_cert.log)
2024-09-02 08:48:31,461 - Parsing config file: cert/cfg/ICVSBContent.cfg
2024-09-02 08:48:31,461 - Parsed config items:
cert-keypair:	cert/key/ICVSBContent.pem
cert-keypair-pwd:	cert/key/ICVSBContent.pwd
load-verify-scheme:	0
crypto-type:	0
aes-ce-id:	0
aes-enc-key:	cert/key/KCEICV.txt
images-table:	cert/Images.txt
nvcounter-val:	1
cert-pkg:	cert/ICVSBContent.crt
2024-09-02 08:48:31,461 - **** Creating Content Certificate ****
2024-09-02 08:48:31,479 - **** Certificate file (context) creation has been completed successfully **** [HMX] Connecting to /dev/tty.usbmodem578D0274271...
[HMX] Connected, press the **RESET** button on your Himax Grove Vision AI Module V2 now
[HMX] Restarted into bootloader. Sending file.
[HMX] Sending 1376 blocks
 ████████████████████████████████████████ 100% | ETA: 0s | 1376/1376
[HMX] Sent all blocks (NAK count: 0)
[HMX] Press **RESET** to start the application...
[HMX] Could not verify application Failed to get a valid response (looking for "Compiler Version") from device. Response was: "
1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms 
waiting input key...
slot flash_offset 0x00000000
New MemDesp himax_sec_SB_image_process PASS
set_memory_s_ns
bl_status = 0x800000, HX_DSP_FLAG 1
bl_status = 0x800000
jump_addr=0x3401f000
1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms 
waiting input key...
slot flash_offset 0x00000000
New MemDesp himax_sec_SB_image_process PASS
set_memory_s_ns
bl_status = 0x800000, HX_DSP_FLAG 1
bl_status = 0x800000
jump_addr=0x3401f000
"
[HMX] Did you press RESET on the board? Am I just not pressing the reset button in time? By the way following your instructions I  kept giving me the following  errors which I want to understand why I keep running into them  when I tried to compile with make but, using source ~/.zshrc  and touch obis.in respectively fixed the error .make: *** [obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/device/WE2_core.o] Error 126 and arm-none-eabi-gcc: error: objs.in: No such file or directory and I am still getting serial output(baud rate: 115200) like this:---- Opened the serial port /dev/tty.usbmodem578D0274271 ----
Ú{rw8v¿W
7ÿ00t0ó¿7ÿÿ\\6tw·2ÃzF¶Ö3FWp4VÎ:Bÿ4v3®V6Tw·²ÃpWp¾·W
7ÿTp0:pVÊ|6VP6rG4zF¶Ö·F×pü¿0{94tVx4ø¿wWs:vP{F00V¿0;t¿4
W^Ö3ÿT·2CF¶p63tK3ý·v8wVx4ø·ÿp6rCF¶p23tKsü6Vq6o·Ïÿ^û{5Fú|ú
---- Closed serial port /dev/tty.usbmodem578D0274271 due to disconnection from the machine ----
---- Opened the serial port /dev/tty.Bluetooth-Incoming-Port ----
---- Closed the serial port /dev/tty.usbmodem578D0274271 ----
---- Opened the serial port /dev/tty.usbmodem578D0274271 ----
wW4pWr¶·W
7ÿ¿V30~¾pVÆ
           \6V
---- Closed the serial port /dev/tty.usbmodem578D0274271 ----
---- Opened the serial port /dev/tty.usbmodem578D0274271 ----
wT4p1pF×p64VpÎ8Bþ"tª§V<ÿ

benclockworks, 2024-09-02:
I checked here https://forum.seeedstudio.com/t/is-my-grove-vision-v2-broken/279835 and I just unboxed my camera a couple of days ago so I don't think its the hardware for it that's going me the 1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms
waiting input key...
slot flash_offset 0x00000000
New MemDesp himax_sec_SB_image_process PASS
set_memory_s_ns
bl_status = 0x800000, HX_DSP_FLAG 1
bl_status = 0x800000
jump_addr=0x3401f000
1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms
waiting input key...
slot flash_offset 0x00000000
New MemDesp himax_sec_SB_image_process PASS
set_memory_s_ns
bl_status = 0x800000, HX_DSP_FLAG 1
bl_status = 0x800000
jump_addr=0x3401f000 response

kris-himax, 2024-09-03:
@benclockworks 
Please follow the readme instruction, and burn the right image file `we2_image_gen_local/output_case1_sec_wlcsp/output.img`.

> :l/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img] done
> exit status code: 0> :l/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img] done
> exit status code: 0
> signed (or raw sec type) file name = sign_formal_cm55m_s_application_p2.img
> AddFlashAlignPadding
> /Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img
> Output image: output_case1_sec_wlcsp/cm55m_s_application.img
> signed (or raw sec type) file name = sign_formal_cm55m_s_application_p2.img
> AddFlashAlignPadding
> /Users/benchikbayor/Seeed_Grove_Vision_AI_Module_V2/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img
> Output image: output_case1_sec_wlcsp/cm55m_s_application.img

You burn the wrong binary image file.

closed 2024-09-16 by kris-himax