HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #53: Exposure issues with OV5647

spamekkel, 2024-09-23:
After flashing the YoloX OD model application with the OV5647 image sensor, I noticed the image is very much underexposed when running the application on a Grove Vision AI V2 board. The driver is found in HIMAX_WE2_Rabboni/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cis_sensor/cis_ov5647 and uses the OV5647_mipi_2lane_640x480.i configuration.

Until now I have not found the reason. Any clue, somebody?

kris-himax, 2024-09-25:
Hi @spamekkel 

Did you try https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od
this repository and use [Himax AI tool](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/releases/download/v1.1/Himax_AI_web_toolkit.zip) to show? 
The readme about Himax AI tool you can reference [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od#send-image-and-meta-data-by-uart).

Thanks,
Kris

spamekkel, 2024-09-25:
Hi Kris,

I just found the cause: the model was not at the correct flash address. Thanks for your help.

Jan.

closed 2024-09-26 by kris-himax