HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #54: The last step of "make" fails on MacOS

yipjustin, 2024-09-24:
Run in the follow problem on my mac (Sonoma 14.6.1).

Would appreciate if there are way to work around it. 

Thank you!

```
EPII_CM55M_APP_S % make
...
arm-none-eabi-gcc -mthumb -mcpu=cortex-m55  -DARMCM55 -mfloat-abi=hard -DALLON_SENSOR_TFLM -DARM_MATH_DSP -DARM_MATH_LOOPUNROLL -DARM_MATH_MVEI -DCM55_BIG -DCMSIS_NN -DCOREV_0P9V -DDBG_MORE -DDEBUG -DEPII_EVB -DETHOSU55 -DETHOSU_ARCH=u55 -DETHOSU_LOG_SEVERITY=ETHOSU_LOG_WARN -DETHOS_U -DEVT_DATAPATH -DHM_COMMON -DIC_PACKAGE_WLCSP65 -DIC_VERSION=30 -DIP_2x2 -DIP_5x5 -DIP_INST_ADCC -DIP_INST_ADCC_HV -DIP_INST_AON_GPIO -DIP_INST_DMA0 -DIP_INST_DMA1 -DIP_INST_DMA2 -DIP_INST_DMA3 -DIP_INST_GPIO_G0 -DIP_INST_GPIO_G1 -DIP_INST_GPIO_G2 -DIP_INST_GPIO_G3 -DIP_INST_I2S_HOST -DIP_INST_I2S_SLAVE -DIP_INST_IIC_HOST -DIP_INST_IIC_HOST_MIPI -DIP_INST_IIC_HOST_SENSOR -DIP_INST_IIIC_SLAVE0 -DIP_INST_IIIC_SLAVE1 -DIP_INST_OSPI_HOST -DIP_INST_PWM0 -DIP_INST_PWM1 -DIP_INST_PWM2 -DIP_INST_QSPI_HOST -DIP_INST_RTC0 -DIP_INST_RTC1 -DIP_INST_RTC2 -DIP_INST_SB_GPIO -DIP_INST_SSPI_HOST -DIP_INST_SSPI_SLAVE -DIP_INST_TIMER0 -DIP_INST_TIMER1 -DIP_INST_TIMER2 -DIP_INST_TIMER3 -DIP_INST_TIMER4 -DIP_INST_TIMER5 -DIP_INST_TIMER6 -DIP_INST_TIMER7 -DIP_INST_TIMER8 -DIP_INST_UART0 -DIP_INST_UART1 -DIP_INST_UART2 -DIP_INST_WDT0 -DIP_INST_WDT1 -DIP_adcc -DIP_adcc_hv -DIP_cdm -DIP_csirx -DIP_csitx -DIP_dma -DIP_dp -DIP_edm -DIP_gpio -DIP_hxautoi2c_mst -DIP_i2s -DIP_i3c_mst -DIP_i3c_slv -DIP_iic -DIP_inp -DIP_inp1bitparser -DIP_inpovparser -DIP_isp -DIP_jpeg -DIP_mb -DIP_mpc -DIP_pdm -DIP_pmu -DIP_ppc -DIP_pwm -DIP_rtc -DIP_scu -DIP_sensorctrl -DIP_spi -DIP_swreg_aon -DIP_swreg_lsc -DIP_timer -DIP_tpg -DIP_u55 -DIP_uart -DIP_vad -DIP_watchdog -DIP_xdma -DLIB_CMSIS_NN -DLIB_COMMON -DLIB_EVENT -DLIB_PWRMGMT -DLIB_QSPI_EEPROM -DLIB_SENSORDP -DLIB_SPI_EEPROM -DLIB_SPI_PTL -DSPI_EEPROM_USE_WB_25Q128JW_INST_ -DTFLM2209_U55TAG2205 -DTF_LITE_MCU_DEBUG_LOG -DTF_LITE_STATIC_MEMORY -DTRUSTZONE -DTRUSTZONE_CFG -DTRUSTZONE_SEC -DTRUSTZONE_SEC_ONLY -D__GNU__ -D__NEWLIB__ -Dseeed -I./CMSIS -I./CMSIS/Driver/Include -I./app -I./app/scenario_app -I./app/scenario_app/allon_sensor_tflm -I./app/scenario_app/allon_sensor_tflm/cis_sensor/cis_ov5647 -I./app/scenario_app/event_handler -I./app/scenario_app/event_handler/evt_datapath -I./board -I./board/epii_evb -I./board/epii_evb/config -I./customer/sec_inc/seeed -I./device -I./device/clib -I./device/inc -I./drivers -I./drivers/2x2 -I./drivers/5x5 -I./drivers/adcc -I./drivers/adcc_hv -I./drivers/cdm -I./drivers/csirx -I./drivers/csitx -I./drivers/dma -I./drivers/dp -I./drivers/edm -I./drivers/gpio -I./drivers/hxautoi2c_mst -I./drivers/i2s -I./drivers/i3c_mst -I./drivers/i3c_slv -I./drivers/iic -I./drivers/inc -I./drivers/inp -I./drivers/inp1bitparser -I./drivers/inpovparser -I./drivers/isp -I./drivers/jpeg -I./drivers/mb -I./drivers/mpc -I./drivers/pdm -I./drivers/pmu -I./drivers/ppc -I./drivers/pwm -I./drivers/rtc -I./drivers/scu -I./drivers/seconly_inc -I./drivers/sensorctrl -I./drivers/spi -I./drivers/swreg_aon -I./drivers/swreg_lsc -I./drivers/timer -I./drivers/tpg -I./drivers/u55 -I./drivers/uart -I./drivers/vad -I./drivers/watchdog -I./drivers/xdma -I./external -I./external/cis/ -I./external/cis/hm_common -I./external/inc -I./interface -I./library/cmsis_nn/cmsis_nn -I./library/cmsis_nn/cmsis_nn/Include -I./library/common -I./library/hxevent -I./library/inference/tflmtag2209_u55tag2205 -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/c -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/core/api -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/kernels -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/kernels/internal -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/kernels/internal/reference -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/micro -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/micro/kernels -I./library/inference/tflmtag2209_u55tag2205/tensorflow/lite/schema -I./library/inference/tflmtag2209_u55tag2205/third_party -I./library/inference/tflmtag2209_u55tag2205/third_party/ethos_u_core_driver/include -I./library/inference/tflmtag2209_u55tag2205/third_party/flatbuffers/include -I./library/inference/tflmtag2209_u55tag2205/third_party/gemmlowp -I./library/inference/tflmtag2209_u55tag2205/third_party/ruy -I./library/pwrmgmt -I./library/pwrmgmt/seconly_inc -I./library/sensordp/inc -I./library/sensordp/internal_inc -I./library/spi_eeprom -I./library/spi_eeprom/eeprom_param -I./library/spi_ptl -I./trustzone/tz_cfg -E -P -xc "app/scenario_app/allon_sensor_tflm/allon_sensor_tflm.ld" -MMD -MF"obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65.d" -MT"obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65.ld" -o "obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65.ld"
Linking               :  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf
arm-none-eabi-gcc -DALLON_SENSOR_TFLM -DARM_MATH_DSP -DARM_MATH_LOOPUNROLL -DARM_MATH_MVEI -DCM55_BIG -DCMSIS_NN -DCOREV_0P9V -DDBG_MORE -DDEBUG -DEPII_EVB -DETHOSU55 -DETHOSU_ARCH=u55 -DETHOSU_LOG_SEVERITY=ETHOSU_LOG_WARN -DETHOS_U -DEVT_DATAPATH -DHM_COMMON -DIC_PACKAGE_WLCSP65 -DIC_VERSION=30 -DIP_2x2 -DIP_5x5 -DIP_INST_ADCC -DIP_INST_ADCC_HV -DIP_INST_AON_GPIO -DIP_INST_DMA0 -DIP_INST_DMA1 -DIP_INST_DMA2 -DIP_INST_DMA3 -DIP_INST_GPIO_G0 -DIP_INST_GPIO_G1 -DIP_INST_GPIO_G2 -DIP_INST_GPIO_G3 -DIP_INST_I2S_HOST -DIP_INST_I2S_SLAVE -DIP_INST_IIC_HOST -DIP_INST_IIC_HOST_MIPI -DIP_INST_IIC_HOST_SENSOR -DIP_INST_IIIC_SLAVE0 -DIP_INST_IIIC_SLAVE1 -DIP_INST_OSPI_HOST -DIP_INST_PWM0 -DIP_INST_PWM1 -DIP_INST_PWM2 -DIP_INST_QSPI_HOST -DIP_INST_RTC0 -DIP_INST_RTC1 -DIP_INST_RTC2 -DIP_INST_SB_GPIO -DIP_INST_SSPI_HOST -DIP_INST_SSPI_SLAVE -DIP_INST_TIMER0 -DIP_INST_TIMER1 -DIP_INST_TIMER2 -DIP_INST_TIMER3 -DIP_INST_TIMER4 -DIP_INST_TIMER5 -DIP_INST_TIMER6 -DIP_INST_TIMER7 -DIP_INST_TIMER8 -DIP_INST_UART0 -DIP_INST_UART1 -DIP_INST_UART2 -DIP_INST_WDT0 -DIP_INST_WDT1 -DIP_adcc -DIP_adcc_hv -DIP_cdm -DIP_csirx -DIP_csitx -DIP_dma -DIP_dp -DIP_edm -DIP_gpio -DIP_hxautoi2c_mst -DIP_i2s -DIP_i3c_mst -DIP_i3c_slv -DIP_iic -DIP_inp -DIP_inp1bitparser -DIP_inpovparser -DIP_isp -DIP_jpeg -DIP_mb -DIP_mpc -DIP_pdm -DIP_pmu -DIP_ppc -DIP_pwm -DIP_rtc -DIP_scu -DIP_sensorctrl -DIP_spi -DIP_swreg_aon -DIP_swreg_lsc -DIP_timer -DIP_tpg -DIP_u55 -DIP_uart -DIP_vad -DIP_watchdog -DIP_xdma -DLIB_CMSIS_NN -DLIB_COMMON -DLIB_EVENT -DLIB_PWRMGMT -DLIB_QSPI_EEPROM -DLIB_SENSORDP -DLIB_SPI_EEPROM -DLIB_SPI_PTL -DSPI_EEPROM_USE_WB_25Q128JW_INST_ -DTFLM2209_U55TAG2205 -DTF_LITE_MCU_DEBUG_LOG -DTF_LITE_STATIC_MEMORY -DTRUSTZONE -DTRUSTZONE_CFG -DTRUSTZONE_SEC -DTRUSTZONE_SEC_ONLY -D__GNU__ -D__NEWLIB__ -Dseeed -L"./linker_script/" -L"./linker_script/gcc" -L"./libs" -Xlinker --gc-sections -Xlinker -print-memory-usage -Xlinker --sort-section=alignment -Xlinker --cref -mthumb -mcpu=cortex-m55  -DARMCM55 -mfloat-abi=hard -L"./linker_script/gcc/" -L"obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65" -Xlinker --cmse-implib  -Wl,-M,-Map=obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.map -specs=nano.specs -T obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65.ld  @objs.in -Wl,--start-group  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libtrustzone_cfg.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libdriver.a   obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libcommon.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libhxevent.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libpwrmgmt.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libsensordp.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/lib_spi_eeprom.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/lib_spi_ptl.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/lib_cmsis_nn.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libextdevice.a    obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libcustomer.a  obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/libtflmtag2209_u55tag2205_cmsisnn_gnu.a     -lm -lc_nano -lgcc -lstdc++_nano -Wl,--end-group -o obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf
arm-none-eabi-gcc: error: objs.in: No such file or directory
make: *** [obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/EPII_CM55M_gnu_epii_evb_WLCSP65_s.elf] Error 1
```

kris-himax, 2024-09-25:
Hi @yipjustin 

You should use GNU make not BSD version make.

You can check the make version by following command.
```
make --version
```
[image]

If it is not GNU make, you should download it by following command.
```
brew install make
```
After installation, you can access it with the command `gmake` to avoid conflicts with the default `make`.

So, you should create an alias in your shell configuration file (like .bash_profile or .zshrc):
```
alias make='gmake'
```

Thanks,
Kris

yipjustin, 2024-09-25:
Kris,

Thanks and resolved after upgrading to GNU Make 4.3.

The error occurred when I am using GNU Make 3.81:

```
(seeed) yipjustin@yipjustin-mac EPII_CM55M_APP_S % make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
This program built for i386-apple-darwin11.3.0
```

closed 2024-09-25 by yipjustin