HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #61: ei_standalone_inferencing_camera sample project does not compile (Error: missing file)

bebemanso, 2024-10-23:
Steps to reproduce:
1. Set `APP_TYPE: ei_standalone_inferencing_camera`
2. Building firmware as per README.md:
- `make clean`
- `make` 

Error:
```
app/scenario_app/ei_standalone_inferencing_camera/ei-model/edge-impulse-sdk/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_f16.c: fatal error: cannot open obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/app/scenario_app/ei_standalone_inferencing_camera/ei-model/edge-impulse-sdk/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_f16.su for writing: No such file or directory
compilation terminated.
make: *** [options/rules.mk:296: obj_epii_evb_icv30_bdv10/gnu_epii_evb_WLCSP65/app/scenario_app/ei_standalone_inferencing_camera/ei-model/edge-impulse-sdk/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_f16.o] Error 1
```

kris-himax, 2024-10-23:
Hi @bebemanso ,

I built this app at Linux environment is fine. Is it built at Windows environment?

Hi @mmajchrzycki 

Please check about this issue. I built this app at Linux environment is fine, but face the same error at Windows environment.

Thanks,
Kris

bebemanso, 2024-10-23:
Hello @kris-himax ,

Yes, I tried building the project at Windows environment.

mmajchrzycki, 2024-10-23:
@bebemanso @kris-himax 
I can confirm the same issue on my Windows machine. It looks like some kind of Windows-related corner case. I'm investigating the root cause. I tried to disable the stack usage checking (this is where the *.su files come from), but then I'm getting another errors related to *.d files.
It looks like we are hitting some Windows OS limitation (path length, number of files or something).
As a workaround, are you able to build the firmware on a Linux VM or using Docker?

bebemanso, 2024-10-23:
@mmajchrzycki

Yes, I was able to build the firmware using Ubuntu. No errors just as @kris-himax said.