HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #62: No connection via SWD

kostalazeps, 2024-10-24:
Hello Himax Team.
My image is larger than 1MB and I thought about using SWD to flash the image. Sadly I get errors like these

```
connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/coresight/dap.py", line 286, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/probe/cmsis_dap_probe.py", line 542, in read_dp
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferFaultError: Memory transfer fault
```

I've tried to connect via openocd and a small script:
```
# Your JTAG adapter config
script interface/cmsis-dap.cfg

transport select swd
adapter_khz 100

swd newdap chip cpu -enable
dap create chip.dap -chain-position chip.cpu
target create chip.cpu cortex_m -dap chip.dap

init
dap info
shutdown
```
This usually should also provide some insight into the connected devices. I have to assume that SWD is somehow not working. I've access to a J-Link which is booting in CMSIS-DAP mode and als a picoprobe which is flashed with the CMSIS-DAP firmware. Both are not able to detect the Grove Vision module.
SWDIO is connected to D2 and SWCLK is connected to clock, like specified in the manual.
I've checked other issues but they don't really seem to explain which step was actually helping. With a Traveo II controller the CMSIS-DAP is actually working.
I've also tried to use 
```
/* Init Arm SWD interface pin mux to PB6, PB7, PB8 (nR, CLK, DIO)*/
void swd_pinmux_cfg(SCU_PINMUX_CFG_T *pinmux_cfg)
{
    pinmux_cfg->pin_pb6 = SCU_PB6_PINMUX_SRSTN;   /*!< pin PB6*/
    pinmux_cfg->pin_pb7 = SCU_PB7_PINMUX_SWCLK_0; /*!< pin PB7*/
    pinmux_cfg->pin_pb8 = SCU_PB8_PINMUX_SWDIO_0; /*!< pin PB8*/
}
```
as opposed to just use the bootloader. I have no SWD connection.

Any ideas?

stevehuang82, 2024-10-25:
Hi @kostalazeps,

When using the swd debugger, the hardware requirement is a CMSIS-DAP compliant debugger(ex. NXP MCU-Link Debug Probe). We ever tried connecting to the Grove Vision AI V2 using the JLink debugger, but it didn't work. Sorry about it.

When using bootloader uart to upgrade firmware, the size of the firmware image cannot be larger than 1MB. Does your code include model files? If so, you can put the model file into flash memory.

kostalazeps, 2024-10-25:
Hello @stevehuang82,
thank you for this very fast response. I'm a little bit confused. I've access to a J-Link which I'm booting in CMSIS-DAP mode and I also have access to a picoprobe which is an open CMSIS-DAP solution. If the only Debugger supported is the NXP one, that should be written somewhere. Maybe a list of supported devices so one can directly buy the correct one.

Considering the solution for putting the model in flash memory. Which is the preferred way of doing that?
Is there a tool available for easier writing to flash?
Can the bootloader put data at random locations?
Do I have to write an application which receives the data over UART to write them to flash manually?
I'm asking to avoid the effort of creating something which already might exist. :-)

Kind regards,
André Zeps

stevehuang82, 2024-10-25:
Hi @kostalazeps,

We have some examples of putting model in the flash memory. For example, [tflm_fd_fm](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/README.md), [tflm_yolov8_od](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/README.md) and [tflm_yolov8_pose](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_pose/README.md)
.
Please refer [tflm_fd_fm](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/README.md) app for example.
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm#readme

https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/FAQ.md#2-can-we-decide-where-to-place-the-model-for-the-tflm_fd_fm-example-in-flash

Below is the relevant code to put the model into flash memory.
(1) Init XIP mode
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/tflm_fd_fm.c#L663
```
 hx_lib_spi_eeprom_open(USE_DW_SPI_MST_Q);
 hx_lib_spi_eeprom_enable_XIP(USE_DW_SPI_MST_Q, true, FLASH_QUAD, true);
```

(2) Define model address here
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/common_config.h#L18
```
#define FACE_DECTECT_FLASH_ADDR         (BASE_ADDR_FLASH1_R_ALIAS+0x200000)
#define FACE_MESH_FLASH_ADDR            (BASE_ADDR_FLASH1_R_ALIAS+0x280000)
#define IRIS_LANDMARKS_FLASH_ADDR       (BASE_ADDR_FLASH1_R_ALIAS+0x32A000)
BASE_ADDR_FLASH1_R_ALIAS(0x3A000000) is flash starting address for MCU
```

(3) Get model from flash
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm/cvapp_fd_fm.cpp#L1050
```
static const tflite::Model*model = tflite::GetModel((const void *)fd_model_addr);
static const tflite::Model*FM_model = tflite::GetModel((const void *)fm_model_addr);
static const tflite::Model*IL_model = tflite::GetModel((const void *)il_model_addr);
```

(4) Program model into flash
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm#readme

Position of model on flash is defined at ~/tflm_fd_fm/common_config.h

python3 xmodem/xmodem_send.py --port=[your COM number] --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local/output_case1_sec_wlcsp/output.img --model="model_zoo/tflm_fd_fm/0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo/tflm_fd_fm/1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo/tflm_fd_fm/2_il_0x32A000.tflite 0x32A000 0x00000"

```
# example:
# python3 xmodem/xmodem_send.py --port=/dev/ttyACM0 --baudrate=921600 --protocol=xmodem --file=we2_image_gen_local/output_case1_sec_wlcsp/output.img --model="model_zoo/tflm_fd_fm/0_fd_0x200000.tflite 0x200000 0x00000" --model="model_zoo/tflm_fd_fm/1_fm_0x280000.tflite 0x280000 0x00000"  --model="model_zoo/tflm_fd_fm/2_il_0x32A000.tflite 0x32A000 0x00000"
```

Kind regards,
Steve

kostalazeps, 2024-10-25:
Thank you a lot for this very elaborated and thorough answer. It gives me also much more insight into what the xmodem utility can do. Maybe I should have read a little bit more into the many README files in this repo. You can close this ticket if you want. However, I still consider it to be not very correct that a J Link in CMSIS-DAP mode is not capable of connecting to the target.

kostalazeps, 2024-11-27:
@stevehuang82 I might need to reopen this.
We've now bought exactly the debugger shown in the picture here:
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/swd_debugging

```
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/coresight/dap.py", line 467, in _connect
    connector.connect()
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/coresight/dap.py", line 257, in connect
    self._idr = self.read_idr()
                ^^^^^^^^^^^^^^^
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/coresight/dap.py", line 286, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/andre/GIT/seeed_grove_vision_ai_module_v2/swd_debugging/pyocd/venv/lib/python3.12/site-packages/pyocd/probe/cmsis_dap_probe.py", line 542, in read_dp
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
```

Neither on Windows, nor in a virtual machine running Debian running on Windows, I have success.
I've measured the pins D2 and TXD using an oscilloscope and there is communication occuring.

Do I need to update something? A bootloader?

The output of the HX83192 is this here:
`1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms
waiting input key
Set X-modem flag = Yes

slot flash_offset 0x00100000
jump_addr=0x3401f000
Compiler Version: ARM CLANG, Clang 13.0.0 (ssh://ds-gerrit/armcompiler/llvm-project 1f5770d6f72ee4eba2159092bbf4cbb819be323a)
set_IP_secure done
flash type[2], flash size[5]
slot FlashOffset 0x00000000
Image max size 0x00100000
------------------------------------------------------------
[0] Reboot system
[1] Xmodem download and burn FW image
------------------------------------------------------------
`
I'm keeping the serial port open so the device doesn't reset. Still for some reason, the communication is not working.

kostalazeps, 2024-11-27:
The debugger is recognized as "NXP Semiconductors MCU-LINK (r0FF) CMSIS-DAP V3.148". I've updated the firmware using NXP linkserver just to be sure that it is a recent firmware.

stevehuang82, 2024-11-28:
Hi @kostalazeps,
Since you are using a Windows development environment, please follow the attachment to set up your SWD flash programming environment.
[Windows_ARM_Open_Development_System_20241128.pdf](https://github.com/user-attachments/files/17943398/Windows_ARM_Open_Development_System_20241128.pdf)

kostalazeps, 2024-11-28:
I'm confused. I've now ensured that it is also the same python version.

> pyocd --version    
0.34.3.dev0+dirty

>python --version                                                      
Python 3.10.5

Still, the problem persists. I've also switched from the flash script to VS Code and used the "EPII_CM55M_APP_S Launch" target.

`
Traceback (most recent call last):
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\__main__.py", line 161, in run
    status = cmd.invoke()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\subcommands\gdbserver_cmd.py", line 190, in invoke
    with session:
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\core\session.py", line 391, in __enter__
    self.open()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\core\session.py", line 529, in open
    self._board.init()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\board\board.py", line 139, in init
    self.target.init()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\core\soc_target.py", line 147, in init
    seq.invoke()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\utility\sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\utility\sequencer.py", line 208, in invoke
    resultSequence = call()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\coresight\dap.py", line 443, in _connect
    connector.connect()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\coresight\dap.py", line 252, in connect
    self._idr = self.read_idr()
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\coresight\dap.py", line 281, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "C:\Projekte\f08312-edcu\tests\radar\venv3\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 443, in read_dp
    raise self._convert_exception(error) from error
pyocd.core.exceptions.TransferError: Unexpected ACK value (5) returned by probe
`

The problem is still the same.
Just to be sure, here is a photo of my setup.
[image]

I've connected GND, SWDIO and SWCLK. Is it possible that this is not enough?

stevehuang82, 2024-11-29:
Please try connecting the MCU-LINK VTref pin to the Grove Vision AI V2 3V3 pin.

kostalazeps, 2024-11-29:
I did. This doesn't change anything.

stevehuang82, 2024-12-02:
Can you put WE2 into firmware upgrade mode and have the console output as follows (only execute the bootloader and not the application)? At this time try to connect to WE2 through SWD.

```
1st BL Modem Build DATE=Nov 30 2023, 0x0002000b
Please input any key to enter X-Modem mode in 100 ms
waiting input key
Set X-modem flag = Yes

slot flash_offset 0x00100000
jump_addr=0x3401f000
Compiler Version: ARM CLANG, Clang 13.0.0 (ssh://ds-gerrit/armcompiler/llvm-project 1f5770d6f72ee4eba2159092bbf4cbb819be323a)
set_IP_secure done
flash type[2], flash size[5]
slot FlashOffset 0x00000000
Image max size 0x00100000
------------------------------------------------------------
[0] Reboot system
[1] Xmodem download and burn FW image
------------------------------------------------------------
```

kostalazeps, 2024-12-02:
I did that. It is also written here in the readme of the github project and the PDF you've provided as well.
I've kept the "a" key pressed on the terminal and pressed and released the RST button. Sometimes it's just booting normally, probably because the repeated key is not in time, but when it stays in that screen I just leave it that way.
I also leave the serial terminal open as closing it leads to a reset.

I then use VSCode with the "EPII_CM55M_APP_S Launch" target.
Are there any ways to analyze this problem? If it helps I can record the communication with a Saleae Logic Analyzer.
Right now it feels a little bit like searching for a needle in a haystack. :-D

stevehuang82, 2024-12-02:
Please confirm that the SWD interface is connected correctly as shown below.

MCU-LINK (SWCLK) -> SWCLK (Grove Vision AI V2 pin TXD)
MCU-LINK (SWDIO) -> SWDIO (Grove Vision AI V2 pin D2)
MCU-LINK (GND) -> GND (Grove Vision AI V2 pin GND)

Please measure the SWCLK and SWDIO signals. The correct waveforms are as follows.
[image]

abalawender, 2024-12-08:
I'm seeing a similar issue with SWD. Here's a capture from my logic analyzer, in case it helps: 

[image]

stevehuang82, 2024-12-09:
Hi @kostalazeps,

Did you install Himax pyOCD according to page 8 of the [Windows_ARM_Open_Development_System_20241128.pdf](https://github.com/user-attachments/files/17943398/Windows_ARM_Open_Development_System_20241128.pdf) installation guide as shown below?

Install Himax pyOCD for WE2
1. Download and install python3 depend on your platform
https://www.python.org

2. Check python
$ python3 --version
Python 3.10.5

3. Upgrade python pip
$ python3 -m pip install --upgrade pip

4. Download pyocd_hx-0.34.3.dev0+dirty-py3-none-any.whl from
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/swd_debugging/pyocd

5. Uninstall pyOCD and reinstall pyocd_hx-0.34.3.dev0+dirty
$ python3 -m pip uninstall pyocd
$ python3 -m pip uninstall pyocd_hx
$ python3 -m pip install pyocd_hx-0.34.3.dev0+dirty-py3-none-any.whl

6. Check pyOCD
$ pyocd --version
0.34.3.dev0+dirty
$ pyocd list --target
Name     Vendor    Part Number    Families Source
we2        Himax     WE2                 builtin

abalawender, 2024-12-09:
@kostalazeps @stevehuang82 thanks. Yes, I am using the `pyocd_hx` (though it should not matter for the lack of IDCODE response). My setup still does not seem to work. I've tried two sets of wires, two different debuggers, and two different HX6538 boards - no combination works as expected.

@kostalazeps is there anything you've changed other than the physical connection? Flashed a different image perhaps?

kostalazeps, 2024-12-10:
@abalawender The image flashed on the board should not matter as the bootloader is already capable of SWD to flash. The only things I did was the accumulation of the PDF supplied here and the [swd_debugging](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/swd_debugging) manual.

I've noticed that there is not really any output documented on how it might look like. So, the serial connection to the Grove is rather weird as it is able to reset the board. So, I keep the terminal open during SWD.
I reset the board and have a button pressed which is not 0 or 1.
My goal is to have this output at last
`
------------------------------------------------------------
[0] Reboot system
[1] Xmodem download and burn FW image
------------------------------------------------------------
`

Afterwards I can start it in VSCode by just opening the `EPII_CM55M_APP_S` folder and using the launch tasks:
`
"C:/x/venv3/Scripts/pyocd.exe" gdbserver --port 50000 --telnet-port 50002 --target we2 -Oenable_multicore_debug=true -Oconnect_mode='halt' -Ofrequency=3600000 -Opersist=false -Oreset_type='sw_sysresetreq' -Oreset.hold_time=0.1 -Oreset.post_delay=1 -Oreset.halt_timeout=2 -Ostep_into_interrupt=false -Ovector_catch='h' -Oenable_semihosting=false -Osmart_flash=false -Oserve_local_only=true -Odap_swj_enable=true -Odap_swj_use_dormant=false -Odap_protocol=swd
0001787 I Target type is we2 [board]
0001798 I DP IDR = 0x4c013477 (v3 MINDP rev4) [dap]
0001802 I Root Component Class 0x9 ROM table #0 @ 0x000000 (designer=c6a part=7d8) [rom_table]
0001805 I [0]<00002000:SDC-600 class=9 designer=43b:Arm part=9ef devtype=00 archid=0a57 devid=0:0:0> [rom_table]
0001807 I [1]<00004000:APB-AP CS-600 class=9 designer=43b:Arm part=9e2 devtype=00 archid=0a17 devid=0:0:0> [rom_table]
0001809 I [2]<00006000:AHB-AP CS-600 class=9 designer=43b:Arm part=9e3 devtype=00 archid=0a17 devid=0:0:0> [rom_table]
0001811 I [3]<00008000:AHB-AP CS-600 class=9 designer=43b:Arm part=9e3 devtype=00 archid=0a17 devid=0:0:0> [rom_table]
0001813 I [4]<0000a000:??? class=15 designer=43b:Arm part=193> [rom_table]
0001814 I APB4-AP@0x4000 IDR = 0x24770006 (APB4-AP var0 rev2) [ap]
0001817 I AHB5-AP@0x6000 IDR = 0x34770008 (AHB5-AP var0 rev3) [ap]
0001820 I AHB5-AP@0x8000 IDR = 0x34770008 (AHB5-AP var0 rev3) [ap]
0001826 I APB4-AP@0x4000 Class 0x9 ROM table #0 @ 0x000000 (designer=c6a part=7d9) [rom_table]
0001829 I [0]<00001000:TPIU CS-600 class=9 designer=43b:Arm part=9e7 devtype=11 archid=0000 devid=20:0:0> [rom_table]
0001830 I [1]<00002000:CTI CS-600 class=9 designer=43b:Arm part=9ed devtype=14 archid=1a14 devid=1040500:0:0> [rom_table]
0001832 I [2]<00003000:ETB CS-600 class=9 designer=43b:Arm part=9e9 devtype=21 archid=0000 devid=300:1:0> [rom_table]
0001834 I [3]<00004000:ATB Replicator CS-600 class=9 designer=43b:Arm part=9ec devtype=22 archid=0000 devid=32:0:0> [rom_table]
0001836 I [4]<00005000:ATB Funnel CS-600 class=9 designer=43b:Arm part=9eb devtype=12 archid=0000 devid=34:0:0> [rom_table]
0001838 I AHB5-AP@0x6000 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4d2) [rom_table]
0001841 I [0]<e000e000:SCS M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0001842 I [1]<e0001000:DWT M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0001844 I [2]<e0002000:BPU M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0001846 I [3]<e0000000:ITM M55 class=9 designer=43b:Arm part=d22 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0001848 I [5]<e0041000:ETM M55 class=9 designer=43b:Arm part=d22 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001850 I [6]<e0003000:PMU M55 class=9 designer=43b:Arm part=d22 devtype=16 archid=0a06 devid=0:0:0> [rom_table]
0001852 I [7]<e0042000:CTI M55 class=9 designer=43b:Arm part=d22 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0001855 I AHB5-AP@0x8000 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b:Arm part=4d2) [rom_table]
0001857 I [0]<e000e000:SCS M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=2a04 devid=0:0:0> [rom_table]
0001859 I [1]<e0001000:DWT M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a02 devid=0:0:0> [rom_table]
0001861 I [2]<e0002000:BPU M55 class=9 designer=43b:Arm part=d22 devtype=00 archid=1a03 devid=0:0:0> [rom_table]
0001863 I [3]<e0000000:ITM M55 class=9 designer=43b:Arm part=d22 devtype=43 archid=1a01 devid=0:0:0> [rom_table]
0001865 I [5]<e0041000:ETM M55 class=9 designer=43b:Arm part=d22 devtype=13 archid=4a13 devid=0:0:0> [rom_table]
0001867 I [6]<e0003000:PMU M55 class=9 designer=43b:Arm part=d22 devtype=16 archid=0a06 devid=0:0:0> [rom_table]
0001869 I [7]<e0042000:CTI M55 class=9 designer=43b:Arm part=d22 devtype=14 archid=1a14 devid=40800:0:0> [rom_table]
0001873 I CPU core #0 is Cortex-M55 r1p1 (security ext present) [cortex_m_v8m]
0001874 I FPU present: FPv5-D16-M [cortex_m]
0001877 I CPU core #1 is Cortex-M55 r1p1 (security ext present) [cortex_m_v8m]
0001878 I FPU present: FPv5-D16-M [cortex_m]
0001881 I 4 hardware watchpoints [dwt]
0001882 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001886 I 4 hardware watchpoints [dwt]
0001887 I 8 hardware breakpoints, 1 literal comparators [fpb]
0001905 I Semihost server started on port 50002 (core 0) [server]
0002140 I GDB server started on port 50000 (core 0) [gdbserver]
0002145 I Semihost server started on port 50003 (core 1) [server]
[... cut ...]
(venv3) PS > pyocd list         
  #   Probe/Board               Unique ID      Target  
-------------------------------------------------------
  0   SEGGER J-Link CMSIS-DAP   000601003931   n/a

(venv3) PS > pyocd list --target
* Prints lots of stuff but last line is*
  we2                       Himax                   WE2                                     builtin
`

I run all of this under Windows 10 which I thought might be an issue at first...
I've installed this special version of pyocd within a venv.

stevehuang82, 2024-12-19:
Hi @abalawender, @kostalazeps,

Are you still having SWD connection issues? Or we can close this issue?

abalawender, 2024-12-19:
@kostalazeps thank you for the details. I'm glad to see your issue is resolved!

SWD still does not work for me. I'll try to use the (undocumented?) built-in I2C debug device when I have time to play with the board again. You can close this ticket.

closed 2024-12-20 by stevehuang82