HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #63: Model flash address setting in common_config.h

jeremypai, 2024-11-08:
Hi, currently I'm trying to upload more than one model in tflm_yolov8_od scenario app . However, when I coded like tflm_fd_fm, I'm facing an issue about model flash address setting in common_config.h. 

From my knowledge, if I want to upload like 3 models at the same time, I have to make sure that we have enough memory for the models to load. For example, if I have models with size of 100,000 bytes, 200,000 bytes, and 300,000 bytes, I have to set the model flash address something like:
#define MODEL_1 (BASE_ADDR_FLASH1_R_ALIAS+0x200000)  // 100,000 bytes model
#define MODEL_2 (BASE_ADDR_FLASH1_R_ALIAS+0x2186A0)  // 200,000 bytes model
#define MODEL_3 (BASE_ADDR_FLASH1_R_ALIAS+0x2493E0)  // 300,000 bytes model

with 0x200000 reserved memory for firmware image

But it turned out that it's didn't work as I expected, all I got was either failed in flash process or printed like the log provided below.

xmodem_send bin file done!!
xmodem_send bin file result =  True
b'\x06'
b''
b'Do you want to end file transmission and reboot system? (y)'
b'\x001st BL Modem Build DATE=Nov 30 2023, 0x0002000b'
b'Please input any key to enter X-Modem mode in 100 ms'
b'waiting input key...'
b'slot flash_offset 0x00000000'
b'New MemDesp himax_sec_SB_image_process PASS'
b'set_memory_s_ns'
b'bl_status = 0x800000, HX_DSP_FLAG 1'
b'bl_status = 0x800000'
b'jump_addr=0x10000000'
b'Compiler Version: ARM GNU, 13.2.1 20231009'
b'wakeup_event=0x0,WakeupEvt1=0x0, freq=400000000'
b'ori_clk src info, 0x56100030=4077'
b'ori_clk src info, 0x56100034=0'
b'ori_clk src info, 0x56100038=c1a4'
b'clk src info, 0x56100030=4037'
b'clk src info, 0x56100034=0'
b'clk src info, 0x56100038=c1b8'
b'__GNUC'
b'__mm_start_addr__ address: 3401f000'
b'YOLOv8n object detection'
b'TA[3401f000]'
b'[TEST] gray image arena: 3412f000'
b'232'
b'Ethos-U55 device initialised'
b'237'
b'244'
b''
b'Entering BusFault_Handler interrupt!'
b'SCB->CFSR:0x00008200'
b'SCB->BFAR:0x23c214f9'
b'SCB->HFSR:0x00000000'

Can you help me figure out how to set common_config.h to upload customized models?

Thanks in advance!

kris-himax, 2024-11-08:
Hi @jeremypai 

You can reference [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/FAQ.md#2-can-we-decide-where-to-place-the-model-for-the-tflm_fd_fm-example-in-flash).
The model starting address should be 4KB aligned. 
So, your model can be placed like
```
#define MODEL_1 (BASE_ADDR_FLASH1_R_ALIAS+0x200000) // 100,000 bytes model, already aligned
#define MODEL_2 (BASE_ADDR_FLASH1_R_ALIAS+0x219000) // 200,000 bytes model, aligned up
#define MODEL_3 (BASE_ADDR_FLASH1_R_ALIAS+0x24A000) // 300,000 bytes model, aligned up
```

Thanks,
Kris

jeremypai, 2024-11-08:
Thanks for your response, that's help a lot!

closed 2024-11-08 by jeremypai