HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #66: Method to check camera input and print it on the screen

jeremypai, 2024-11-20:
Hi, I'm trying to port my own models on the device and check if those models could work normally on it.

However, after I flashed the firmware and models, I couldn't find the results as I expected.
When I wanted to find if there's any tool that could help me fix this problem, I couldn't find one.

Can you tell me that if there's any tool that could help me get the image from camera sensor, and even print some information such as I can draw some bounding boxes on it and display it on the screen?

By the way, because currently I didn't find any tool that could help me get the image, I tried to encode the test image as a C array and wrote it in a file. Then, I got the following logs... Can you also kindly tell me what's happening in this case?

arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: region RAM overflowed with stack
Memory region         Used Size  Region Size  %age Used
 CM55M_S_APP_ROM:      156196 B       256 KB     59.58%
CM55M_S_APP_DATA:       66420 B       256 KB     25.34%
    CM55M_S_SRAM:          0 GB      1924 KB      0.00%
collect2: error: ld returned 1 exit status

Thanks in advance!

kris-himax, 2024-11-21:
Hi @jeremypai ,
You can reference the [tflm_yolov8_od](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od) example to show image and the bounding box on [Himax AI web toolkit](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/releases/download/v1.1/Himax_AI_web_toolkit.zip) which can reference the instruction guide at [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/README.md#send-image-and-meta-data-by-uart).

For your second question, I don't know how you set your linker script about your data on the SRAM. Maybe you can try the `tflm_yolov8_od` example first.

Thanks,
Kris

jeremypai, 2024-11-21:
Hi @kris-himax,
Thanks for your reply. Just want to check if we're on the same page.

For the question 1:
Is `hx_drv_spi_mst_protocol_write_sp` what's you mean that I can set the bounding box and the image data which would be shown on the screen? By the way, I see DATA_TYPE_META_YOLOV8_OB_DATA, can I define my own data type on structure SPI_CMD_DATA_TYPE?

For the question 2:
Currently I didn't change the default setting on linker script. As the log provided like "region RAM overflowed with stack" I think I should increase the size of CM55M_S_APP_DATA. However, I found that the device would stuck after printing the following logs. Do you have any idea what's happenning?

b'\x001st BL Modem Build DATE=Nov 30 2023, 0x0002000b'
b'Please input any key to enter X-Modem mode in 100 ms'
b'waiting input key...'
b'slot flash_offset 0x00100000'
b'New MemDesp himax_sec_SB_image_process PASS'
b'set_memory_s_ns'
b'bl_status = 0x800000, HX_DSP_FLAG 1'
b'bl_status = 0x800000'
b'jump_addr=0x10000000'

kris-himax, 2024-11-21:
Hi @jeremypai 

1. No, you can set your bbox result at [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cvapp_yolov8n_ob.cpp#L390). No, you can not.
2. It seems like that the stack size is not enough.

Thanks,
Kris

jeremypai, 2024-11-26:
Hi @kris-himax 
How do I decide where to put images on, SRAM or CM55M_S_APP_DATA?
Is there any example codes can guide me to implement it?

kris-himax, 2024-11-28:
Hi @jeremypai 
You can reference the example at [allon_sensor_tflm.ld](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/allon_sensor_tflm/allon_sensor_tflm.ld#L76) which put the g_person_detect_model_data_vela array on SRAM.

Thanks,
Kris

jeremypai, 2024-11-28:
Hi @kris-himax 
Ok I got it. Thanks for your help!