HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #67: Himax AI web toolkit preview result not working

annietllnd, 2024-11-21:
Hi! I've tried using the toolkit to display the running of the model according to the [guide](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/README.md#send-image-and-meta-data-by-uart). I'm using the index.html file (it didn't seem to work in Firefox, but displays are expected in Chrome. So far so good). The rest of the guide works great, and I can verify the flashing part works thanks to the terminal output printing person detection scores after pushing the reset button.

After connecting the board successfully through the UI, the page doesn't update to show the input image (which happens successfully when using the [web hosted version](https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process), selecting the YOLOv5 model for example). Instead, the page looks the same, the device log is empty, and nothing happens on clicking the different blue model use-case tiles. I tried it on both MacOS and Linux.

Any ideas on what to try here? If I can provide any more information, just let me know. 

Thanks in advance, I really appreciate your time 🌟

kris-himax, 2024-11-22:
Hi @annietllnd,
You should build and burn the scenario apps like [tflm_yolov8_od](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od), [tflm_fd_fm](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_fd_fm), [tflm_yolov8_pose](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_pose), or [hello_world_cmsis_cv](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/hello_world_cmsis_cv) which will send the image and result via UART.
The rest of the scenario apps did not include the code to send the image via UART.

Thanks,
Kris

annietllnd, 2024-11-24:
Hi Kris, I'm running the yolov8 object detection use-case, as instructed in the guide. I get confirmation that the board is successfully connected through the UI, but then the view remains to look like [the screenshot](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/images/uart_show_1.png) from the README.md. 

Like mentioned, after burning the image, the person score is printed and seems to hit a 100 when a person appears in front of the camera. 
```
b'SENSORDPLIB_STATUS_XDMA_FRAME_READY 240'
b'write frame result 0, data size=15284,addr=0x340e04e0'
b'invoke pass'
b'person_score:113'
b'EVT event = 10'
b'SENSORDPLIB_STATUS_XDMA_FRAME_READY 241'
b'write frame result 0, data size=15296,addr=0x340e04e0'
b'invoke pass'
b'person_score:112'
b'EVT event = 10'
```

annietllnd, 2024-11-27:
Hi Kris, 
it was a mistake on my end - I hadn't set the `APP_TYPE` in the `makefile`. This does enable the UART feed.

Thanks,
Annie