HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #68: Using a custom model with Grove Vision AI Module V2

AnnaPunzengruber, 2024-12-04:
Hello,

I am trying to use the Grove Vision AI Module V2 with a custom model to detect and count people from an overhead view. I created a dataset in roboflow (about 800 images, 192x192) and trained a yolov8 model using ultralytics. I then exported the model as an int8 quantized tflite file. I also converted the model using ethos-u-vela. The final model has the correct format int8_vela.tflite and is about 3MB in size. 
However, when I upload the model to the Grove Vision AI Module V2 using the [SenseCraft Model Assistant](https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process), I get an "Invoke failed" error. 
I have already contacted technical support and they told me to open a new issue here. Do you have any idea why this error occurs? What can I do to make my model work with Grove Vision AI Module V2?

Kind regards,
Anna

kris-himax, 2024-12-05:
Hi @AnnaPunzengruber 
If you want to use the [SenseCraft Model Assistant](https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process) which is no code platform.
You should use the [model training script script](https://wiki.seeedstudio.com/ModelAssistant_Introduce_Quick_Start/#model-training) which provided by Seeed studio.
If you want to use yolov8 model modified by yourself, maybe you should modify the code. You can reference the [tflm_yolov8_od](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od) example.

Thanks,
Kris

AnnaPunzengruber, 2024-12-10:
Hi @kris-himax,
I adapted my code with [YOLOv8_on_WE2](https://github.com/HimaxWiseEyePlus/YOLOv8_on_WE2) and it worked. 
I think the problem was the way I was converting the .tflite model to vela. 
Thank you for your help!

Kind regards,
Anna