HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #70: Maximum support camera resolution on Grove Vision AI module V2 (Urgent, Seed technical support sent us to you)

serhiparad3, 2024-12-09:
Hello,
We are designing a new product based on Grove Vision AI module V2 and want to use the camera resolution 640x518 and 10-15 frames/second(jpeg).
The camera(IMX462) provides RAW10 format, we need to make colors tuning by ISP and convert to YUV422 or similar format.
Which is teh meximum rsolution that we could handle and compress to jpeg? 
Do you a have a camera drivers porting guide? How to make porting asap ?

serhiparad3, 2024-12-10:
Thank you for the information,
We need to get 640x520 pixels. Can we get in some way 10 frames 640х520? 
For an example read to RAM 640x520, divide on two halfs 640x260 and 640x260, after compress in 2 separate jpeg files ? (half and half photo) 
My understanding, we don't have limitation with reading photo, just with encoding jpeg? For an example, we could read 1024x768 and divide on 4 parts and encoding after ?(4 jpeg files)

stevehuang82, 2024-12-26:
Hi @serhiparad3,

Yes, you are right. We have an allon_jpeg_encode example that demonstrates how to capture a 1280x960 raw image and convert the 1280x960 image into four 640x480 jpeg images. Hope this helps.

https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/tree/main/EPII_CM55M_APP_S/app/scenario_app/allon_jpeg_encode

serhiparad3, 2025-01-21:
Hello,

Another question:
We have the Arducam IMX462 camera, which uses an additional small MCU for communication (to convert the IMX462's I2C protocols to the Arducam protocol).
[Arducam IMX462 Camera](https://www.arducam.com/product/2mp-imx462-color-ultra-low-light-starvis-camera-module-with-141h-wide-angle-m12-lens-for-raspberry-pi/)

The camera streams at 30 frames per second (this cannot be changed as it is not supported In Arducam protocol) with a resolution of 1920×1080 (this could potentially be adjusted). However, we need the output to be 640×480 at 10 frames per second, with the output format in JPEG.

Will the Grove Vision 2 handle this? If yes, could you provide information on how we should configure the camera driver? (We are using the IMX219 driver from the SDK as a reference.)

serhiparad3, 2025-01-23:
Any feedback?

yuriiliudt, 2025-01-29:
Hello,

I would like to understand what values ​​are indicated here:
https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cis_sensor/cis_ov5647/cisdp_sensor.c#L187
What are they for?
Do they need to be changed in any way if a different camera module is used?
`	uint32_t n_preload = 15;
	uint32_t l_header = 4;
	uint32_t l_footer = 2;
`
I didn't find any mention of this in the documentation.