HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #72: arm64 linux version of ./we2_local_image_gen

Gostas, 2024-12-14:
Hi there, I am using a Raspberry Pi 5 and currently getting the error `bash: ./we2_local_image_gen: cannot execute binary file: Exec format error` when trying to create the firmware image.

Could you please add a linux arm64 version of this binary?

Thanks

stevehuang82, 2024-12-17:
Hi @Gostas,

Sorry, we currently have no plans to support the linux arm64 version.

Gostas, 2024-12-18:
Oh okay. I saw that you added support for mac on arm fairly recently. Would it be more difficult to add an executable for linux arm64?

Alternatively, is there some way I can make the firmware image with the provided binary on my Raspberry Pi?

stevehuang82, 2024-12-18:
We do not have a Raspberry Pi environment to build the image generation tool library.
Maybe you could use a Windows, Ubuntu or Mac environment to build the code and generate the firmware image, it should be more efficient.

closed 2024-12-18 by Gostas