HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #73: yolov8 classification support

ThomasBroch, 2024-12-31:
Hi,

I'm working on adding support for yolov8n classification models. Are there, by any chance, examples available for the output processing of a classification model? [Here](https://github.com/Seeed-Studio/sscma-example-we2/blob/da4b23e11a54273f086d533bda193fc945745b12/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cvapp_yolov8n_ob.cpp#L612), the tflm yolov8 od firmware does the post processing for an object detection model output. Thus far I have not been able to find documentation relating to a classification model output and the required post processing steps.

This is what I have now:

```
static void yolov8_cl_post_processing(tflite::MicroInterpreter *static_interpreter, float modelScoreThreshold, std::forward_list<el_class_t> &el_algo)
{
	TfLiteTensor *output = static_interpreter->output(0);
	int num_classes = output->dims->data[1];

	xprintf("Number of classes: %d\n", num_classes);

	float output_scale = ((TfLiteAffineQuantization *)(output->quantization.params))->scale->data[0];
	int output_zeropoint = ((TfLiteAffineQuantization *)(output->quantization.params))->zero_point->data[0];

	float max_score = -1;
	uint16_t maxClassIndex = 0;

	for (int i = 0; i < num_classes; i++)
	{
		int value = output->data.int8[i];
		xprintf("class: %d -  value: %d", i, value);

		float deq_value = ((float)value - (float)output_zeropoint) * output_scale;
		xprintf("class: %d -  deq_value: %d", i, deq_value);
		if (deq_value > max_score)
		{
			max_score = deq_value;
			maxClassIndex = i;
		}
	}

	xprintf("Max score: %d\n", max_score);

	if (max_score >= modelScoreThreshold)
	{
		el_class_t result;
		result.score = static_cast<uint16_t>(max_score * 100); // Convert to percentage
		result.target = maxClassIndex;

		el_algo.emplace_front(result);
	}
}
```

It makes use of the `el_class_t` struct that was already present in the firmware [here](https://github.com/Seeed-Studio/sscma-example-we2/blob/da4b23e11a54273f086d533bda193fc945745b12/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/send_result.h#L75).

kris-himax, 2025-01-02:
Hi @ThomasBroch ,
The Yolov8n is reference from https://github.com/ultralytics/ultralytics.
You can find out the documentation over there.

Thanks,
Kris

ThomasBroch, 2025-01-02:
Hi @kris-himax ,

Unfortunately, ultralytics doesn't provide much info on yolov8 classification. Upon doing some further debugging I think I've found a bug in the output processing that also affects the object detection example firmware.

When running a yolov8n ob model with one output, the example firmware reads the dims data to fetch the number of possible output labels [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/fb3c4c2afd2a98a84e376b18ac7777a1801a7e87/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_od/cvapp_yolov8n_ob.cpp#L419). However, `output->dims->data[1]` appears to always come back as `8`, regardless of how many labels the model actually has. I've tested this with an OB model with 4 labels and one with 10 labels and it always says 8.

Furthermore, when we iterate over `output->data.int8` when running a classification model, we're expecting to see that each value corresponds with a confidence corresponding to that index/label, but we're not seeing anything like that either. We are seeing changes on the output tensor when feeding it different images, but not in a way that we can make sense of.

Could you help us in figuring out how to run a yolov8n classification model and interpret the output? The way the model is initialized does not appear to contain anything object detection specific. Is there anything we're missing when we want to run classification? The classification model we have appears to run just fine, but the output data is not making a lot of sense.

Thnx!

kris-himax, 2025-02-13:
Hi @ThomasBroch ,
You can reference the [yolov8n gender classification example](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_gender_cls/README.md). And the output processing, you can reference [here](https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/EPII_CM55M_APP_S/app/scenario_app/tflm_yolov8_gender_cls/cvapp_yolov8_gender_cls.cpp#L506). 

Thanks,
kris

closed 2025-02-14 by kris-himax