HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #75: How to transfer 4 bytes of data to a register via I2C using hx_drv_cis_setRegTable or hx_drv_cis_set_reg or other method?

yuriiliudt, 2025-01-16:
I need to develop a new camera sensor driver, but in this SDK I can only send 1 byte of data. And there are no driver sources, only precompiled libraries.
Please help me solve this.

stevehuang82, 2025-01-20:
Hi @yuriiliudt,

Below is the source code of the hx_drv_cis_get_register() and hx_drv_cis_set_register() functions that can read and write 1 byte data.
You can refer to these functions and modify them to transfer 4 bytes of data to a register.
You can put these functions into cisdp_sensor.c.

```
#define OV5647_SENSOR_I2CID     (0x36)
#define HX_CIS_I2C_ADDR_BYTE    2  	/**< I2C Master Register address length*/
#define HX_CIS_I2C_DATA_BYTE    1  	/**< I2C Master Register value length*/

HX_CIS_ERROR_E hx_drv_cis_get_register(uint16_t addr, uint8_t *val) {
	uint8_t RegByteCount = 2;
	HX_CIS_ERROR_E status = HX_CIS_NO_ERROR;
	uint8_t rwBuffer[HX_CIS_I2C_DATA_BYTE] = { 0x00 };
	uint8_t regAddr[HX_CIS_I2C_ADDR_BYTE];
	int32_t retI2C = 0;

    SENSORCTRL_MCLKCTRL_E mclkctrl;
	hx_drv_sensorctrl_get_MCLKCtrl(&mclkctrl);
	if(mclkctrl != SENSORCTRL_MCLKCTRL_NONAOS)
	{
		hx_drv_sensorctrl_set_MCLKCtrl(SENSORCTRL_MCLKCTRL_NONAOS);
	}

    //pin mux set to default
	drv_interface_set_sensor_i2cm_pinmux(SCU_SEN_I2CM_SDA_PINMUX_I2CM_SDA_0, SCU_SEN_I2CM_SCL_PINMUX_I2CM_SCL_0);

	*val = 0;

	regAddr[0] = (addr >> 8) & 0xFF;
	regAddr[1] = addr & 0xFF;

    retI2C = hx_drv_i2cm_write_restart_read(HX_CIS_IIC_M_ID, OV5647_SENSOR_I2CID, regAddr, RegByteCount, rwBuffer, HX_CIS_I2C_DATA_BYTE);
    if(retI2C < E_OK)
    {
    	xprintf("HX CIS getRegister I2C fail\r\n");
  	  return HX_CIS_ERROR_I2C;
    }else{
    	status = HX_CIS_NO_ERROR;
    }

    if (HX_CIS_NO_ERROR == status) {
    	*val = rwBuffer[0];
    }else{
    	xprintf("hx_drv_cis_get_reg fail 0x%x\r\n", regAddr);
    }

	return HX_CIS_NO_ERROR;
}

HX_CIS_ERROR_E hx_drv_cis_set_register(uint16_t addr, uint8_t val) {
	uint8_t rwBuffer[HX_CIS_I2C_DATA_BYTE] = { 0x00 };
	uint8_t regLen = HX_CIS_I2C_ADDR_BYTE;
	uint8_t regAddr[HX_CIS_I2C_ADDR_BYTE];
	int32_t retI2C = 0;

    SENSORCTRL_MCLKCTRL_E mclkctrl;
	hx_drv_sensorctrl_get_MCLKCtrl(&mclkctrl);
	if(mclkctrl != SENSORCTRL_MCLKCTRL_NONAOS)
	{
		hx_drv_sensorctrl_set_MCLKCtrl(SENSORCTRL_MCLKCTRL_NONAOS);
	}

    //pin mux set to default
	drv_interface_set_sensor_i2cm_pinmux(SCU_SEN_I2CM_SDA_PINMUX_I2CM_SDA_0, SCU_SEN_I2CM_SCL_PINMUX_I2CM_SCL_0);

	regAddr[0] = (addr >> 8) & 0xFF;
	regAddr[1] = addr & 0xFF;

	rwBuffer[0] = val;

	retI2C = hx_drv_i2cm_write_data(HX_CIS_IIC_M_ID, OV5647_SENSOR_I2CID, regAddr, regLen, rwBuffer, HX_CIS_I2C_DATA_BYTE);
	if(retI2C < E_OK)
	{
		xprintf("set_reg I2C fail\r\n");
		return retI2C;
	}

	return HX_CIS_NO_ERROR;
}
```

yuriiliudt, 2025-01-20:
Thank you. I will try your solution.

closed 2025-03-19 by stevehuang82