HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #78: Adjusting "white balance"

yuriiliudt, 2025-02-04:
I have a camera without hardware support for "white balance". Is it possible to set up "white balance" automatically in the SDK?

yuriiliudt, 2025-02-11:
[image]
[image]
I got an image from a custom camera, but it's more green.
How can I set the white balance in the SDK?
In your example with the OV5647 camera, the camera registers are set to automatic white balance.
But in my camera, this is not possible.
Is there a solution to this issue?

yousefbilal, 2025-02-21:
I am facing the same issue with IMX219 camera sensor

closed 2025-11-27 by kris-himax