HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #79: Micro SD and YOLOV8 OD example model work

Vladimirkvvel, 2025-02-09:
For start was used an example YOLOV8 OD
To add work with micro SD was modified file
`app\scenario_app\tflm_yolov8_od\tflm_yolov8_od.mk `
there is a part of code

```
LIB_SEL = pwrmgmt sensordp tflmtag2209_u55tag2205 spi_ptl hxevent img_proc

##
# middleware support feature
# Add new middleware here
# The source code should be loacted in ~\middleware\{mid_name}\
##
MID_SEL = fatfs 

#FATFS_PORT_LIST = mmc_spi flash ram
FATFS_PORT_LIST = mmc_spi
#FATFS_PORT_LIST = flash
#FATFS_PORT_LIST = ram

CMSIS_DRIVERS_LIST = SPI
```
Other part of micro SD implementation got from example. File syesem works fine

From `app\scenario_app\tflm_yolov8_od\tflm_yolov8_od.c` was removed work with SPI and SPI EEPROM as like as:

```
//#include "spi_eeprom_comm.h"
//pinmux_init();
// hx_lib_spi_eeprom_open(USE_DW_SPI_MST_Q);
//hx_lib_spi_eeprom_open_speed(USE_DW_SPI_MST_Q, 6000000);
// hx_lib_spi_eeprom_enable_XIP(USE_DW_SPI_MST_Q, true, FLASH_QUAD, true);

```

Seems like after that object recognization stoped work
in file `app\scenario_app\tflm_yolov8_od\cvapp_yolov8n_ob.cpp` 

int cv_yolov8n_ob_i```
nit(bool security_enable, bool privilege_enable, uint32_t model_addr) {
	int ercode = 0;

	//set memory allocation to tensor_arena
	tensor_arena = mm_reserve_align(tensor_arena_size,0x20); //1mb
	xprintf("TA[%x]\r\n",tensor_arena);

	if(_arm_npu_init(security_enable, privilege_enable)!=0)
		return -1;

	if(model_addr != 0) {
		static const tflite::Model*yolov8n_ob_model = tflite::GetModel((const void *)model_addr);

		if (yolov8n_ob_model->version() != TFLITE_SCHEMA_VERSION) {
			xprintf(
				"[ERROR] yolov8n_ob_model's schema version %d is not equal "
				"to supported version %d\n",
				yolov8n_ob_model->version(), TFLITE_SCHEMA_VERSION);
			return -1;
		}
```
The `yolov8n_ob_model->version()` is `0`

As I found in the module external memory connected to QSPI interface
Is it possible to make work Object detection and micro SD together?

kris-himax, 2025-02-13:
Hi @Vladimirkvvel,

It is possible to make work Object detection and micro SD.
You don't need to remove work with SPI and SPI EEPROM at app\scenario_app\tflm_yolov8_od\tflm_yolov8_od.c .
But you need to disable `#define GROVE_VISION_AI_II`, to let pinux to be switch to be following setting.
```
	pinmux_cfg->pin_pb2 = SCU_PB2_PINMUX_SPI_M_DO_1;        /*!< pin PB2*/
	pinmux_cfg->pin_pb3 = SCU_PB3_PINMUX_SPI_M_DI_1;        /*!< pin PB3*/
	pinmux_cfg->pin_pb4 = SCU_PB4_PINMUX_SPI_M_SCLK_1;      /*!< pin PB4*/
	pinmux_cfg->pin_pb5 = SCU_PB5_PINMUX_SPI_M_CS_1;        /*!< pin PB5*/
```
You also need to read model from flash by opening XIP mode.
```
hx_lib_spi_eeprom_open(USE_DW_SPI_MST_Q);

hx_lib_spi_eeprom_enable_XIP(USE_DW_SPI_MST_Q, true, FLASH_QUAD, true);
```

Thanks,
Kris

Vladimirkvvel, 2025-02-13:
Hello @kris-himax 

Thanks 

The issue was solved after SDK update.
All parts of Object detection by default and SD added as like as in example

Best regards,
Vladimir

closed 2025-02-13 by Vladimirkvvel