HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2 #90: HX6538-A Custom Board Development Issue. Can't Flash by SWD

vienkmt, 2025-09-19:
Dear Support Team,

We are developing an AI product based on your HX6538-A chip and require assistance with the following issues:

- While SeeedStudio's design uses the HX6538-A01TWA, our new design implements the HX6538-A04TLDG package. For the flash memory, we are using a GigaDevice alternative to the Winbond chip. (GD25LQ128ESIGR)

- Current Issue: I cannot flash firmware to our custom development board. I am using an ST-Link V2 with PB7 connected to SWCLK and PB8 connected to SWDIO. When attempting to flash via SWD, I encounter the following errors:

`vienkmt@vienkmt:~/Desktop/s/swd_debugging/swdflash$ ./burn.sh output.img
./burn.sh
output.img
swdflash.py version 1.0.6
===== start =====
Traceback (most recent call last):
  File "swdflash.py", line 844, in <module>
    main()
  File "swdflash.py", line 708, in main
    session.open()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 439, in _connect
    probe_conn.connect(self._protocol)
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/probe/stlink_probe.py", line 159, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/home/vienkmt/.local/lib/python3.8/site-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
vienkmt@vienkmt:~/Desktop/s/swd_debugging/swdflash$ 
`

- Could you please review our schematic (attached) and provide feedback?  Due to limited documentation available online, your technical support would be greatly appreciated. Thank you for your assistance.

[New-Schematic.pdf](https://github.com/user-attachments/files/22426792/New-Schematic.pdf)

[image]

vienkmt, 2025-09-24:
@stevehuang82 
Thank you, I am placing an order for MCU-Link and will attempt again later. I have tried J-Link, PicoProbe but they are not compatible. I will update information soon.

stevehuang82, 2025-09-26:
Hi @vienkmt 

Before using SWD for flash programming, please make sure to force the Seeed Grove Vision AI Module V2 into debugging mode, so that it does not run the application stored in flash. Kindly follow the steps below.
 
1. Plug in Seeed Grove Vision AI Module V2
2. Open Minicom, setup serial port and COM Port name-> connect to Grove Vision AI Module V2. (Please reference https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/blob/main/README.md)
3. Hold down any key on the keyboard (except the Enter key) and press the reset button to reset Grove Vision AI Module V2 and the startup options will be displayed.

[image]

[image]

4. Do not enter [0] or [1], just stop here.
5. Start to program flash.

stevehuang82, 2025-09-26:
Hi @vienkmt 

Could you take a photo of the entire setup, including the NXP MCU-Link Debug Probe, SWD connector, and Grove Vision AI module, for my reference?
Please make sure that SWCLK, SWDIO, and GND are all securely connected.
You can also refer to the following link, which includes information about common SWD connection issues.

No connection via SWD : https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/issues/62

vienkmt, 2025-09-26:
Thanks for your help but still nothing new. I will try to order another CMIS-DAP board. Please wait for my update.

vienkmt, 2025-09-26:
Thanks @stevehuang82  pls provide on Github or share to my email vienkmt@hotmail.com. The board can be purchased, nothing difficult, just solve the problem.
Many thanks again

vienkmt, 2025-09-29:
hi @stevehuang82 
Can you provide instructions for uploading bootloader via I2C today? I apologize for the rush, but we really need it. The project has been delayed for several months. 
Thank you very much.

stevehuang82, 2025-10-03:
Hi @vienkmt 

Have you successfully programmed the flash using SWD or the I2C ISP tool?

vienkmt, 2025-10-16:
hi @stevehuang82 
I can flash by I2C on Grove but can't on my board. I will debug hardware on my board.
Thanks so much for your support 

[image]

stevehuang82, 2025-10-16:
Hi @vienkmt 

1. You can use the following command to read the HX6538-A chip ID and confirm that the device is powered on.
$ ./isp_tool_single -i  
read address: 0x46100ff0 [0x8538000d]

2. Since the GD25LQ128ESIGR is not a fixed quad mode device, external pull-up resistors must be added to IO2 and IO3.

vienkmt, 2025-10-16:
Hi,
Perhaps there's an issue with my hardware, I'll check to see what the problem is. As shown below, the command runs successfully with Grove AI but it's not quite right on my PCB.

[image]

vienkmt, 2025-10-24:
hi @stevehuang82 
Update October 24th: 
I bought a commercially available HX6538 board to test SWD flashing, and it worked immediately. However, when I tried again, I still cannot flash the bootloader to GROVE AI via SWD. The good news is I can still flash Grove via I2C. 

What I need to do now is recheck my hardware design. Thank you for your help. This issue can be closed
This is commercially HX6538 board
https://www.elecrow.com/grc-ai-add-on-for-crowpanel-on-hx6538.html

[image]